project-chip/connectedhomeip #15384: How to pair a device using qr-code and commission wifi/thread

caipiblack, 2022-02-21:
#### Problem

I'm looking for the correct way to pair with qr-code **and commission** a device using the Matter stack.

Before this PR (https://github.com/project-chip/connectedhomeip/pull/12909) the chip-tool example was doing the commissioning step when the callback `OnPairingComplete()` was triggered. 

Why this example has been removed ? How is the correct way to do it ?

In my opinion, when `OnPairingComplete()` is triggered, we have to:

- Look in which network type we want to comission the device (thread/wifi, depending on device capabilities and the network available in the fabric) 
- Ask to the user for the network credentials (or read from memory)
- Use the `Network Commissioning Cluster` to commission the device

Is this correct ?

cecille, 2022-02-25:
Please see https://github.com/project-chip/connectedhomeip/pull/14429. This got a little stale as we prioritized some other work in the lead up to the last branch, but I should get back to it either late today or early next week.

caipiblack, 2022-03-09:
@cecille : I also have more questions:

What is the final objective of all the commissioning callbacks ?
- Is it just to get informations about the commissioning status ? Or is there something to do in the callbacks during the commissioning ?

Also, is there any plans to improve these callbacks ? (regroup them and add more status)

Currently If I do the commissioning of a thread device from the application, we have to see the Matter logs to know what happens, 

I think it should be good to implement more states in the `OnStatusUpdate()` callback so we can use it to display "commissioning status" on the final application.

For example:
- Searching the device (BLE, SoftAP, OnNetwork)
- Device discovered (BLE, SoftAP, OnNetwork)
- Waiting for the device in the end network (should be called when we know the device is going to connect on final network)
- Device discovered in the end network

Or better: add a status for each commissioning steps in `chip::Controller::DevicePairingDelegate::Status` ! 

```
    enum Status : uint8_t
    {
        SecurePairingSuccess = 0,
        SecurePairingFailed,
    };

    /////////// callbacks from DevicePairingDelegate /////////
    void OnStatusUpdate(chip::Controller::DevicePairingDelegate::Status status) override;
    void OnPairingComplete(CHIP_ERROR error) override;
    void OnCommissioningComplete(NodeId deviceId, CHIP_ERROR error) override;
```

cecille, 2022-03-10:
Callbacks cleanup issue: https://github.com/project-chip/connectedhomeip/issues/15612

Please feel free to add status callback updates if you will find them helpful. You can also achieve this by using a custom commissioner delegate.

cecille, 2022-03-29:
https://github.com/project-chip/connectedhomeip/pull/14429 handles this issue. Callbacks is tracked in another issue, so closing this.

closed 2022-03-29 by cecille