project-chip/connectedhomeip #15386: [TC-MC-3.10]Media Input Cluster - ShowInputStatus and HideInputStatus Commands are fails to list the active input status info in

Rajashreekalmane, 2022-02-21:
Test-Case - [TC-MC-3.10] Show and Hide Input Status Verification
 
#### Problem
ShowInputStatus and  HideInputStatus Commands are send  to DUT to verify the active input status info, but these commands are fails to list the input status information.
 
* expected behavior: it should display the status info on the device with successful response (Value 0).
* actual behavior: Displaying only the success response. 

Test-Plan reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/media.adoc#tc-mc-3-10-show-and-hide-input-status-verification

Spec reference:
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/media/MediaInput.adoc#inputinfo

PFA log below:
[[TC-MC-3.10].txt](https://github.com/project-chip/connectedhomeip/files/8109925/TC-MC-3.10.txt)

lazarkov, 2022-04-14:
Can you please update the test plans and remove the part "device should show the input status"?

franck-apple, 2022-04-14:
> Can you please update the test plans and remove the part "device should show the input status"?

if this is the only problem, please open an issue in the test-plan repo https://github.com/CHIP-Specifications/chip-test-plans/issues and close this current issue.

closed 2022-04-18 by Rajashreekalmane