project-chip/connectedhomeip #15423: Handle querying a non-existent provider for OTA

carol-apple, 2022-02-22:
#### Problem
When a `QueryImage` is sent to a non-existent provider on the fabric, at least on Darwin, the DNS lookup seems to never return. This could put OTA in a state where another query cannot be performed (because the current state is not idle for the next possible query).

 #### Proposed Solution
Set some kind of timer to check if a query has been outstanding

isiu-apple, 2022-03-10:
Looks like on OTA-R Linux, DNSServiceResolve() never returns on invalid provider node and therefore the QueryImage hangs.  Whereas on ESP32/M5Stack, DNSServiceResolve() will time out.

https://github.com/project-chip/connectedhomeip/blob/master/src/lib/dnssd/ActiveResolveAttempts.cpp#L197

carol-apple, 2022-03-10:
This will be fixed by https://github.com/project-chip/connectedhomeip/pull/16061

isiu-apple, 2022-03-11:
The `DNSServiceResolve() `not calling the error callback after a timeout issue has been resolved with https://github.com/project-chip/connectedhomeip/pull/15934.  With this fix, OTA-R is now able to handle sending `QueryImage` to a non-existent provider on the fabric, recover from the error, and successfully handle subsequent `QueryImage` commands sent to a valid provider.

```
[1647038463595] [46159:1105665] CHIP: [EM] Received message of type 0x10 with protocolId (0, 0) and MessageCounter:4924637 on exchange 49455r
[1647038463595] [46159:1105665] CHIP: [EM] Found matching exchange: 49455r, Delegate: 0x0
[1647038463595] [46159:1105665] CHIP: [EM] Rxd Ack; Removing MessageCounter:13195649 from Retrans Table on exchange 49455r
[1647038463595] [46159:1105665] CHIP: [EM] Removed CHIP MessageCounter:13195649 from RetransTable on exchange 49455r
[1647038463794] [46159:1105665] CHIP: [DIS] Checking node lookup status after 200 ms
[1647038473594] [46159:1105664] CHIP: [DIS] Checking node lookup status after 10000 ms
[1647038473594] [46159:1105664] CHIP: [DIS] Operational discovery failed for 0x000000000000AAAA: ../../../examples/ota-requestor-app/linux/third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:178: CHIP Error 0x00000032: Timeo
[1647038473594] [46159:1105664] CHIP: [SWU] //is: OTARequestor::OnConnectionFailure
[1647038473594] [46159:1105664] CHIP: [SWU] Failed to connect to node 0x000000000000AAAA: ../../../examples/ota-requestor-app/linux/third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:178: CHIP Error 0x00000032: Timeout
[1647038473594] [46159:1105664] CHIP: [SWU] //is: OTARequestor::RecordErrorUpdateState
[1647038473594] [46159:1105664] CHIP: [EVL] LogEvent event number: 0x0000000000190003 priority: 1, endpoint id:  0x0 cluster id: 0x0000_002A event id: 0x2 Sys timestamp: 0x0000000009E96452
[1647038473595] [46159:1105664] CHIP: [SWU] //is: OTARequestor::RecordNewUpdateState
[1647038473595] [46159:1105664] CHIP: [DMG] Endpoint 0, Cluster 0x0000_002A update version to 4d6f35c3
[1647038473595] [46159:1105664] CHIP: [EVL] LogEvent event number: 0x0000000000190004 priority: 1, endpoint id:  0x0 cluster id: 0x0000_002A event id: 0x0 Sys timestamp: 0x0000000009E96452
[1647038473595] [46159:1105664] CHIP: [SWU] //is: GenericOTARequestorDriver::StartDefaultProviderTimer
[1647038473595] [46159:1105664] CHIP: [SWU] Starting the Default Provider timer, timeout: 86400 seconds
[1647038473595] [46159:1105664] CHIP: [SWU] //is: GenericOTARequestorDriver::UpdateNotFound
[1647038473595] [46159:1105664] CHIP: [SWU] //is: GenericOTARequestorDriver::DetermineProviderLocation
[1647038473595] [46159:1105664] CHIP: [SWU] No suitable OTA Provider candidate found
[1647038473595] [46159:1105664] CHIP: [SWU] UpdateNotFound, not scheduling further retries
[1647038473595] [46159:1105664] CHIP: [DIS] Discovery does not require any more timeouts
```

closed 2022-03-11 by isiu-apple