project-chip/connectedhomeip #15426: Investigate and re-enable the OTA CI test

selissia, 2022-02-22:
#### Problem
OTA CI tests were disabled in https://github.com/project-chip/connectedhomeip/pull/15424 because of instability, possibly related to mDNS. Slack discussion:  https://csamembers.slack.com/archives/G014G30SVV0/p1645552481568699

Need to investigate what's happening and eventually re-enable the CI.

 #### Proposed Solution

mykrupp, 2022-02-22:
When Linux fails, the requestor does not commission. 

When Darwin fails, send ota announce provider command fails

Sometimes github also fails to clone some repos

mykrupp, 2022-02-28:
https://github.com/project-chip/connectedhomeip/pull/15648

mykrupp, 2022-03-04:
https://github.com/project-chip/connectedhomeip/pull/15659

mykrupp, 2022-03-04:
The linux test fails about 20% of the time. 
When the linux test fails, it is because the requestor does not commission. 
The requestor doesn't commission because chip-tool erroneously finds the port corresponding to the provider instead of the requestor in these cases.

**Some logs when commissioning succeeds**
CHIP:TOO: Discovered Device: fe80::4402:baff:fee7:5200:5540  when commisioning provider
CHIP:TOO: Discovered Device: fe80::4402:baff:fee7:5200:5560  when commissioning requestor

**Some logs when commissioning fails**
CHIP:TOO: Discovered Device: fe80::4402:baff:fee7:5200:5540  when commissioning provider
CHIP:TOO: Discovered Device: fe80::4402:baff:fee7:5200:5540  when commissioning requestor

As it can be seen, the requestor is given the port corresponding to the provider, causing commissioning to fail. 

It needs to be determined why MDNS sometimes resolves the incorrect port for the requestor.

mykrupp, 2022-03-04:
The incorrect port is picked up here - https://github.com/project-chip/connectedhomeip/blob/3ee4f4dfb04fef383a83fec2e38ab4f1411d72c4/examples/chip-tool/commands/pairing/PairingCommand.cpp#L203

mykrupp, 2022-03-04:
**Chip-tool and requestor logs upon failure:**
[chip-tool-commission-requestor.txt](https://github.com/project-chip/connectedhomeip/files/8188039/chip-tool-commission-requestor.txt)
[requestor-log.txt](https://github.com/project-chip/connectedhomeip/files/8188037/requestor-log.txt)

selissia, 2022-03-04:
The Requestor UDP port is always set to 5560 by using the the command line option `--secured-device-port 5560` .

mykrupp, 2022-03-10:
I tested a few solutions suggested by other Matter developers 

1. Generate a random UDP_PORT and DISCRIMINATOR to make the code hermetic.
2. put rm -r /tmp/chip_* back in the code.

This did not fix the unstableness of the test.

bzbarsky-apple, 2022-05-21:
@carol-apple Is this still an issue given the YAML coverage you added?