project-chip/connectedhomeip #15460: [TC-MC-1.2] Global attributes - Keypad Input Cluster, ServerGeneratedCommandList attribute does not returns a list.

anaamrit, 2022-02-23:
TC : TH reads the ServerGeneratedCommandList attribute from the DUT
Expected : Verify that the DUT response provides a list of supported commands, which for this cluster should be a list containing the value 0.

TC : https://github.com/CHIP-Specifications/chip-test-plans/blob/jsena-media-17Feb22/src/cluster/media.adoc#tc-mc-1-2-global-attributes-keypad-input-cluster-dut-as-server

PFA Log
[ServerGeneratedCommandList.txt](https://github.com/project-chip/connectedhomeip/files/8125217/ServerGeneratedCommandList.txt)

joesena-amz, 2022-02-24:
This list should contain the id for the SendKeyResponse as per https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/Data-Model.adoc#1331-clientgeneratedcommandlist

lazarkov, 2022-04-20:
[1650492581496] [94864:5804842] CHIP: [TOO] Endpoint: 1 Cluster: 0x0000_0509 Attribute 0x0000_FFFB DataVersion: 2125935106
[1650492581496] [94864:5804842] CHIP: [TOO]   AttributeList: 5 entries
[1650492581497] [94864:5804842] CHIP: [TOO]     [1]: 65528
[1650492581497] [94864:5804842] CHIP: [TOO]     [2]: 65529
[1650492581497] [94864:5804842] CHIP: [TOO]     [3]: 65531
[1650492581497] [94864:5804842] CHIP: [TOO]     [4]: 65532
[1650492581497] [94864:5804842] CHIP: [TOO]     [5]: 65533
[1650492581497] [94864:5804842] CHIP: [EM] Sending Standalone Ack for MessageCounter:7189025 on exchange 24465i
[1650492581497] [94864:5804842] CHIP: [IN] Prepared secure message 0x700008050480 to 0x0000000000000001 (1)  of type 0x10 and protocolId (0, 0) on exchange 24465i with MessageCounter:7488008.
[1650492581497] [94864:5804842] CHIP: [IN] Sending encrypted msg 0x700008050480 with MessageCounter:7488008 to 0x0000000000000001 (1) at monotonic time: 0000000049634F1C msec
[1650492581497] [94864:5804842] CHIP: [EM] Flushed pending ack for MessageCounter:7189025 on exchange 24465i
[1650492581497] [94864:5804838] CHIP: [CTL] Shutting down the commissioner
[1650492581497] [94864:5804838] CHIP: [CTL] Shutting down the controller
[1650492581497] [94864:5804838] CHIP: [CTL] Shutting down the commissioner
[1650492581497] [94864:5804838] CHIP: [CTL] Shutting down the controller
[1650492581497] [94864:5804838] CHIP: [IN] Expiring all connections for fabric 1!!
[1650492581497] [94864:5804838] CHIP: [CTL] Shutting down the commissioner
[1650492581497] [94864:5804838] CHIP: [CTL] Shutting down the controller
[1650492581497] [94864:5804838] CHIP: [IN] Expiring all connections for fabric 2!!
[1650492581497] [94864:5804838] CHIP: [CTL] Shutting down the commissioner
[1650492581497] [94864:5804838] CHIP: [CTL] Shutting down the controller
[1650492581497] [94864:5804838] CHIP: [IN] Expiring all connections for fabric 3!!
[1650492581497] [94864:5804838] CHIP: [CTL] Shutting down the System State, this will teardown the CHIP Stack
[1650492581497] [94864:5804838] CHIP: [DMG] IM WH moving to [Uninitialized]
[1650492581497] [94864:5804838] CHIP: [DMG] IM WH moving to [Uninitialized]
[1650492581497] [94864:5804838] CHIP: [DMG] IM WH moving to [Uninitialized]
[1650492581497] [94864:5804838] CHIP: [DMG] IM WH moving to [Uninitialized]
[1650492581497] [94864:5804838] CHIP: [DMG] All ReadHandler-s are clean, clear GlobalDirtySet
[1650492581497] [94864:5804838] CHIP: [DL] Inet Layer shutdown
[1650492581497] [94864:5804838] CHIP: [DL] System Layer shutdown

`➜  connectedhomeip git:(master) ✗ out/debug/standalone/chip-tool keypadinput read attribute-list 1 1`

Confirmed working now.

closed 2022-04-20 by lazarkov