project-chip/connectedhomeip #15461: [TC-MC-1.2] Global attributes - Keypad Input Cluster, Can Not read EventList Attribute.

anaamrit, 2022-02-23:
TC : TH reads the EventList attribute from the DUT

Expected : DUT response provides a list of supported events, which for this cluster should be an empty list.

TC Reference : https://github.com/CHIP-Specifications/chip-test-plans/blob/jsena-media-17Feb22/src/cluster/media.adoc#tc-mc-1-2-global-attributes-keypad-input-cluster-dut-as-server

Actual Output in chip tool

[~/Desktop/Project_Matter]$ chip-tool keypadinput read
[1645628205929] [18685:156153] CHIP: [TOO] Missing attribute name
Usage:
  /Users/anaamrit/Desktop/Project_Matter/connectedhomeip/out/debug/standalone/chip-tool keypadinput read attribute-name [param1 param2 ...]

  +-------------------------------------------------------------------------------------+
  | Attributes:                                                                         |
  +-------------------------------------------------------------------------------------+
  | * server-generated-command-list                                                     |
  | * client-generated-command-list                                                     |
  | * attribute-list                                                                    |
  | * feature-map                                                                       |
  | * cluster-revision                                                                  |
  +-------------------------------------------------------------------------------------+
[1645628205930] [18685:156153] CHIP: [TOO] Run command failure: ../../examples/chip-tool/commands/common/Commands.cpp:120: Error 0x0000002F
[~/Desktop/Project_Matter]$

joesena-amz, 2022-02-24:
Looks like EventList may not be in v1 scope? https://github.com/project-chip/connectedhomeip/issues/15011

bzbarsky-apple, 2022-05-21:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/15011

closed 2022-05-21 by bzbarsky-apple