project-chip/connectedhomeip #15478: iOS chiptool app continues to resubcribe for reporting after fabric/accessory removed

kean-apple, 2022-02-23:
#### Problem
chiptool continue to resubcribe for reporting after fabric/accessory removed

SHA: 9f09e5290e5ce97f446158ac83aaf1942fc4f540
Steps:
1. Pair M5 board (all clusters app) using iOS chiptool
2. Verify able to control/toggle light
3. Go to temperature sensor menu, set reporting: Min: 1s, Max: 2s, Delta: 1C - verify M5 board sends report and App sees temp change updates
4. Go to unpair devices - remove device/M5 board

In step 4, looks like chiptool tries to resubscribe for reports even though the fabric/M5 board was removed. This stops if you kill the chiptool app (background/foreground app doesn't stop it)

M5 board: starts receiving data even though its removed
E (100507) chip[IN]: Data received on an unknown session (LSID=2). Dropping it!
E (100807) chip[IN]: Data received on an unknown session (LSID=2). Dropping it!
E (101117) chip[IN]: Data received on an unknown session (LSID=2). Dropping it!
E (117097) chip[IN]: Data received on an unknown session (LSID=2). Dropping it!
E (122107) chip[IN]: Data received on an unknown session (LSID=2). Dropping it!
E (122417) chip[IN]: Data received on an unknown session (LSID=2). Dropping it!

//looks like CHIP tool is trying to resubscribe for reports even though Mt5 board was removed
2022-02-23 09:01:47.548850-0800  localhost CHIPTool[10226]: (CHIP) [com.zigbee.chip:all] ð¢ [1645635707548] [10226:360844] CHIP: [EM] Retransmitting MessageCounter:15354423 on exchange 13151i Send Cnt 1
2022-02-23 09:01:47.549025-0800  localhost CHIPTool[10226]: (CHIP) [com.zigbee.chip:all] ðµ [1645635707549] [10226:360844] CHIP: [IN] Sending encrypted msg 0x10e03e868 with MessageCounter:15354423 to 0x0000000000000003 (1) at monotonic time: 1645635707549 msec
2022-02-23 09:01:47.848742-0800  localhost CHIPTool[10226]: (CHIP) [com.zigbee.chip:all] ð¢ [1645635707848] [10226:360844] CHIP: [EM] Retransmitting MessageCounter:15354423 on exchange 13151i Send Cnt 2
2022-02-23 09:01:47.848904-0800  localhost CHIPTool[10226]: (CHIP) [com.zigbee.chip:all] ðµ [1645635707848] [10226:360844] CHIP: [IN] Sending encrypted msg 0x10e03e868 with MessageCounter:15354423 to 0x0000000000000003 (1) at monotonic time: 1645635707848 msec
2022-02-23 09:01:48.148562-0800  localhost CHIPTool[10226]: (CHIP) [com.zigbee.chip:all] ð´ [1645635708148] [10226:360862] CHIP: [EM] Failed to Send CHIP MessageCounter:15354423 on exchange 13151i sendCount: 3 max retries: 3
2022-02-23 09:01:54.247341-0800  localhost CHIPTool[10226]: (CHIP) [com.zigbee.chip:all] ðµ [1645635714247] [10226:360862] CHIP: [DMG] Time out! failed to receive report data from Exchange: 13151i
2022-02-23 09:01:54.247613-0800  localhost CHIPTool[10226]: (CHIP) [com.zigbee.chip:all] ðµ [1645635714247] [10226:360862] CHIP: [DMG] Computing Resubscribe policy: attempts 3, max wait time 20000 ms, selected wait time 15091 ms
2022-02-23 09:01:54.247754-0800  localhost CHIPTool[10226]: (CHIP) [com.zigbee.chip:all] ðµ [1645635714247] [10226:360862] CHIP: [DMG] Will try to Resubscribe at retry index 3 after 15091ms
2022-02-23 09:01:54.247856-0800  localhost CHIPTool[10226]: (CHIP) [com.zigbee.chip:all] ð¢ [1645635714247] [10226:360862] CHIP: [DMG] MoveToState ReadClient[0x282f5c5a0]: Moving to [      Idle]
2022-02-23 09:02:09.339217-0800  localhost CHIPTool[10226]: (CHIP) [com.zigbee.chip:all] ðµ [1645635729339] [10226:360862] CHIP: [IN] Prepared secure message 0x10e03e868 to 0x0000000000000003 (1)  of type 0x3

 #### Proposed Solution
<suggested fix, suggested enhancement>
[chiptool-resubcribes-reporting-after-removing-accessory-M5-logs.txt](https://github.com/project-chip/connectedhomeip/files/8126589/chiptool-resubcribes-reporting-after-removing-accessory-M5-logs.txt)

kean-apple, 2022-03-05:
still see this issue on 8e8f4197ed040a6edff042e2c0635fabedd5a7da

vinay-apple, 2022-03-08:
Same issue seen with latest SHA: c0373bbc244894909fb0acf55c73444e14088208

[chiptool-tries-to-resubscribe-after-unpair-M5-logs.txt](https://github.com/project-chip/connectedhomeip/files/8202118/chiptool-tries-to-resubscribe-after-unpair-M5-logs.txt)

[chiptool-tries-to-resubscribe-after-unpair-chiptool-logs.txt](https://github.com/project-chip/connectedhomeip/files/8202117/chiptool-tries-to-resubscribe-after-unpair-chiptool-logs.txt)

kean-apple, 2022-03-19:
Not seen since wild card reporting changes went in

closed 2022-03-19 by kean-apple