project-chip/connectedhomeip #15509: Event-List is not implemented in chip-tool but present in Spec.

Rajashreekalmane, 2022-02-24:
#### Problem
Some clusters have Event-List in Spec but doesn't implement in chip-tool.
Level Control cluster
illuminance Measurement Cluster
Identify cluster
Temperature Measurement Cluster
Thermostat Cluster
Thermostat User Configuration Cluster

Chip-tool Screenshot:
[image]

somu1710, 2022-03-17:
Event list is not implemented in chip-tool but present in spec  for these clusters also
1. on/off
2. modeselect
3. pressure measurement
4. occupancy sensing 
5. Relativehumidity
6. pump configuration and control
7. windowcovering
8. power source
9. power source configuration

sowmyassp, 2022-03-18:
Event list is not implemented in chip-tool but present in spec for these clusters also
1. Boolean State Cluster
2. Flow Measurement Cluster
3. Color Control Cluster
4. Bridged Actions Cluster

Survensa, 2022-04-12:
Need event list implementation for descriptor cluster

cjandhyala, 2022-04-13:
@andy31415 @vivien-apple @bzbarsky-apple Along with EventList, Global Elements talks about other fields like "Accepted CommandList" and "Generate dCommandList" , Are these two are also out of scope ? Ref Section 7.13 from Spec

[image]

bzbarsky-apple, 2022-04-13:
Everything there except EventList is implemented.

Rajashreekalmane, 2022-04-14:
Closing this issue in SDK as @cjandhyala @vivien-apple and @andy31415 as suggested above.

closed 2022-04-14 by Rajashreekalmane