project-chip/connectedhomeip #15517: ytgrrrrrrgftgtyyhuhnujyjyhkhkrefsoikfdvopklfdvcplgvbh[;p[]'ghhjg'[]/

AntonGrey, 2022-02-24:
In some case of using MoveToLevel command transition time can be significant increased. Also affected commands: move-to-level, move and step commands.

 * expected behavior: Current level should be changed in Transition time
 * actual behavior: Level transition time significantly more than expected.
 * system configuration: MIPS platform, TE7.5 Matter sdk version.
 * DUT:  bridge device with dimmable light as bridged node. 
 * steps to reproduce: 
 1. Read current level 
 ```
❯ ./chip-tool levelcontrol read current-level 12344321 1
...
[1645539204.306910][376642:376647] CHIP:TOO: LevelControl.CurrentLevel response: 1
...
```
2. Move to current level 200 with transition time 3 seconds
 ```
./chip-tool levelcontrol move-to-level 200 30 0 0 12344321 1
```
Actually transition time was about 20 seconds instead of 3 as expected.
```
[1645539436.399512][2168:2170] CHIP:ZCL: Event: move from 1
[1645539436.399584][2168:2170] CHIP:ZCL:  to 2 
[1645539436.399623][2168:2170] CHIP:ZCL: (diff +1)
...
[1645539456.914243][2168:2170] CHIP:ZCL: Event: move from 199
[1645539456.914325][2168:2170] CHIP:ZCL:  to 200 
[1645539456.914365][2168:2170] CHIP:ZCL: (diff +1)
```
However, if the value by which the level changes per second is not very large, then there is no significant difference between the expected and actual time. For example:
Move to current level 30 from 10 with transition time 5 seconds.
```
❯ ./chip-tool levelcontrol read current-level 12344321 1
[1645540260.309285][378493:378498] CHIP:TOO: LevelControl.CurrentLevel response: 10

❯ ./chip-tool levelcontrol move-to-level 30 50 0 0 12344321 1
[1645540275.279694][378719:378732] CHIP:DMG: Received Command Response Status for Endpoint=1 Cluster=0x0000_0008 Command=0x0000_0000 Status=0x0
[1645540275.279699][378719:378732] CHIP:TOO: Default Success Response
```

In this case expected transition time approximately equal to actual transition time.
```
[1645540275.528804][2168:2170] CHIP:ZCL: Event: move from 10
[1645540275.528871][2168:2170] CHIP:ZCL:  to 11 
[1645540275.529802][2168:2170] CHIP:ZCL: (diff +1)
...
[1645540281.093490][2168:2170] CHIP:ZCL: Event: move from 29
[1645540281.093557][2168:2170] CHIP:ZCL:  to 30 
[1645540281.093598][2168:2170] CHIP:ZCL: (diff +1)
```
# Another reproduce example
TC-LVL-4.1 can  be used to reproduce the issue:
 1. Read current level 
 ```
❯ ./chip-tool levelcontrol read current-level 12344321 1
...
[1645537629.771916][374405:374410] CHIP:TOO: LevelControl.CurrentLevel response: 1
...
```
 2.  Let's send a move command to DUT with the move mode field set to 0x00 (move up) and the rate field set to 0x20 (32 units per second)
```
❯ ./chip-tool levelcontrol move 0 32 0 0 12344321 1
```
Note that move level time was about 26 seconds instead of 8 as expected.
Part of a DUT log:
```
[1645537666.489862][2168:2170] CHIP:ZCL: Event: move from 1
[1645537666.489930][2168:2170] CHIP:ZCL:  to 2 
[1645537666.489969][2168:2170] CHIP:ZCL: (diff +1)
...
[1645537692.843721][2168:2170] CHIP:ZCL: Event: move from 254
[1645537692.843790][2168:2170] CHIP:ZCL:  to 255 
[1645537692.843830][2168:2170] CHIP:ZCL: (diff +1)
```

bzbarsky-apple, 2022-02-24:
Is there a lower limit to how often tasks can run on this platform?  200 levels in 3s corresponds to 15ms timers.  The observed behavior is what would happen if a timer can't be set for less than 100ms for some reason.  It's hard to tell whether that's an issue here or not without seeing more of the log, not just the first and last bits.

Rajashreekalmane, 2022-09-28:
@AntonGrey Test-plan has been updated and verification steps also updated accordingly, Could you please re-verify the TC's, If the issue is resolved  can you please close the issue.

closed 2023-05-01 by stale[bot]