project-chip/connectedhomeip #15521: EFR32 Pairing failing

rgoliver, 2022-02-24:
#### Problem
EFR32 lock and lighting app are failing commissioning using chip-tool:
`./chip-tool pairing ble-thread $node_id $hex 73141520 3840`

With:
```[1645717402.264884][6420:6425] CHIP:SC: Failed to verify peer's MAC. This can happen when setup code is incorrect.```
 
It looks like the pairing code in the README is no longer correct, using the chip default pairing code gets further but still fails:
`./chip-tool pairing ble-thread $node_id $hex 20202021 3840`

With:
```
[1645717446.471660][6440:6445] CHIP:IN: Clearing BLE pending packets.
[1645717475.890354][6440:6445] CHIP:DMG: Time out! failed to receive invoke command response from Exchange: 27189i
[1645717475.890476][6440:6445] CHIP:CTL: Received failure response ../../examples/chip-tool/third_party/connectedhomeip/src/app/CommandSender.cpp:211: CHIP Error 0x00000032: Timeout

[1645717475.890564][6440:6445] CHIP:CTL: Failed to perform commissioning step 16
```

jmartinez-silabs, 2022-02-24:
All examples now have to use the default 20202021 until we can provide the device with a spake salt and verifier to match the desired non-default setup pin code. We will update documentation, that part slipped.

To my knowledge commissioning work with `20202021` (was working yesterday when I did it). Can you provide some more logs?

I guess I can retest it with today's master.

mkardous-silabs, 2022-02-24:
I just ran the commiossioning on master and it works.

mkardous-silabs, 2022-02-24:
Might just need a factory reset and reboot for the controller/otbr.
Both MG12 and MG24 are working on my side

closed 2022-02-24 by rgoliver