project-chip/connectedhomeip #15524: FLASH error while building the example for CC2652R7

prashanthumbre, 2022-02-24:
# Problem
I am trying to build the Matter node example for CC2652R7 on the latest commit.
Example: 
1. lock-app example: [https://github.com/project-chip/connectedhomeip/tree/master/examples/lock-app/cc13x2x7_26x2x7]
2. pump-app example: [https://github.com/project-chip/connectedhomeip/tree/master/examples/pump-app/cc13x2x7_26x2x7]
Both examples show the same issues while building.

## **1st Issue while building:**
After performing the command: `ninja -C out/debug` getting below error:

```
ninja: Entering directory `out/debug'
[322/822] c++ obj/third_party/connectedhomeip/src...cc13x2_26x2/cc13x2_26x2.PlatformManagerImpl.cpp.o
FAILED: obj/third_party/connectedhomeip/src/platform/cc13x2_26x2/cc13x2_26x2.PlatformManagerImpl.cpp.o 
arm-none-eabi-g++ -MMD -MF obj/third_party/connectedhomeip/src/platform/cc13x2_26x2/cc13x2_26x2.PlatformManagerImpl.cpp.o.d -march=armv7e-m -mcpu=cortex-m4 -mabi=aapcs -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb -Os -g2 -fno-common -ffunction-sections -fdata-sections --specs=nosys.specs --specs=nano.specs -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -Wall -Werror -Wstack-usage=8192 -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-unknown-warning-option -Wno-maybe-uninitialized -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unused -Wno-psabi -Wno-cast-function-type -fdiagnostics-color -fno-strict-aliasing -Wno-comment @gen/sysconfig/ti_ble_app_config.opt @gen/sysconfig/ti_build_config.opt -Wno-conversion -Wno-comment @/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/ble5stack/config/build_components.opt @/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/ble5stack/config/factory_config.opt -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DNDEBUG -DCHIP_HAVE_CONFIG_H=1 -DDeviceFamily_CC13X2X7_CC26X2X7 -DBUILD_RELEASE=1 -DDeviceFamily_CC26X2X7 -DINCLUDE_BGET_STATS -DUSE_DMM -DFLASH_ROM_BUILD -DNVOCMP_FREERTOS_MUTEX=1 -DUSEOT -DONE_BLE_LIB_SIZE_OPTIMIZATION -DICALL_EVENTS -DICALL_JT -DICALL_LITE -DICALL_MAX_NUM_ENTITIES=6 -DICALL_MAX_NUM_TASKS=3 -DICALL_STACK0_ADDR -DPOWER_SAVING -DSTACK_LIBRARY -DTBM_ACTIVE_ITEMS_ONLY -DNPI_USE_UART -DNPI_FLOW_CTRL=0 -DOSAL_CBTIMER_NUM_TASKS=1 -DUSE_ICALL -DBLE_START -DCC26XX -DCC26X2 -DCC26X2R1_LAUNCHXL -DFREERTOS -DNVOCMP_POSIX_MUTEX -DOPENTHREAD_CONFIG_FILE=\<OpenThreadConfig.h\> -DOPENTHREAD_FTD=1 -DOPENTHREAD_MTD=0 -DOPENTHREAD_RADIO=0 -DMBEDTLS_CONFIG_FILE=\"cc13x2_26x2-mbedtls-config.h\" -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE=\"openthread-core-cc13x2_26x2-config.h\" -DOPENTHREAD_CORE_CONFIG_PLATFORM_CHECK_FILE=\"openthread-core-cc13x2_26x2-config-check.h\" -DPACKAGE_NAME=\"OPENTHREAD\" -DPACKAGE_VERSION=\"1.0.0\" -I../../third_party/connectedhomeip/src/include -I../../third_party/connectedhomeip/src -Igen/include -I../../third_party/connectedhomeip/zzz_generated/app-common -I../../third_party/connectedhomeip/src/lwip/cc13x2_26x2 -I../../third_party/connectedhomeip/src/lwip/freertos -I../../third_party/connectedhomeip/third_party/lwip/repo/lwip/src/include -I../../third_party/connectedhomeip/examples/lock-app/cc13x2x7_26x2x7/main/include -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/kernel/freertos/builds/cc13x2x7_26x2x7/release -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/third_party/CMSIS/Include -I../../third_party/connectedhomeip/third_party/mbedtls/repo/include -I../../third_party/connectedhomeip/src/platform/cc13x2_26x2 -I../../third_party/connectedhomeip/third_party/freertos/repo/portable/GCC/ARM_CM4F -I../../third_party/connectedhomeip/third_party/freertos/repo/include -Igen/sysconfig -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/dmm/apps/common/freertos -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/dmm/apps/common/freertos/itm -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/dmm/thread/platform -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/dmm -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/dmm/apps -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/dmm/apps/common/thread -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/dmm/apps/common/thread/source/activity -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/common/nv -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/devices/cc13x2x7_cc26x2x7 -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/devices/cc13x2x7_cc26x2x7/inc -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/devices/cc13x2x7_cc26x2x7/rf_patches -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/devices/cc13x2x7_cc26x2x7/driverlib -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/ble5stack/common/cc26xx/npi/stack -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/ble5stack/common/cc26xx/rcosc -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/ble5stack/controller/cc26xx/inc -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/ble5stack/inc -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/ble5stack/rom -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/ble5stack/common/cc26xx -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/ble5stack/common/cc26xx/freertos -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/ble5stack/icall/inc -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/ble5stack/icall/src -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/ble5stack/hal/src/target/_common -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/ble5stack/hal/src/target/_common/cc26xx -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/ble5stack/hal/src/inc -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/ble5stack/heapmgr -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/ble5stack/profiles/dev_info -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/ble5stack/profiles/simple_profile -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/ble5stack/icall/src/inc -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/ble5stack/osal/src/inc -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/common/cc26xx -I../../third_party/connectedhomeip/third_party/openthread/repo/examples/platforms -I../../third_party/connectedhomeip/third_party/openthread/repo/src/core -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/source/ti/devices/cc13x2x7_26x2x7 -I/home/prashanthumbre/ti/simplelink_cc13xx_cc26xx_sdk_5_30_03_01_eng/examples/rtos/LP_CC2652R7/thread/cli_ftd -I../../third_party/connectedhomeip/third_party/openthread/repo/include -I../../third_party/connectedhomeip/src/platform/cc13x2_26x2/crypto -I../../third_party/connectedhomeip/third_party/openthread/repo/src -I../../third_party/connectedhomeip/third_party/nlassert/repo/include -I../../third_party/connectedhomeip/third_party/nlio/repo/include -c ../../third_party/connectedhomeip/src/platform/cc13x2_26x2/PlatformManagerImpl.cpp -o obj/third_party/connectedhomeip/src/platform/cc13x2_26x2/cc13x2_26x2.PlatformManagerImpl.cpp.o
In file included from ../../third_party/connectedhomeip/src/platform/cc13x2_26x2/PlatformManagerImpl.cpp:31:
../../third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.cpp:116:2: error: #error Must either set INCLUDE_xSemaphoreGetMutexHolder = 1 in FreeRTOSConfig.h or set chip_stack_lock_tracking = "none" in Matter gn configuration.
  116 | #error Must either set INCLUDE_xSemaphoreGetMutexHolder = 1 in FreeRTOSConfig.h or set chip_stack_lock_tracking = "none" in Matter gn configuration.
      |  ^~~~~
[... cut ...] 40,00000000000b0ecf]
/home/prashanthumbre/ti/connectedhomeip/.environment/cipd/pigweed/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 0 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

This seems like FLASH related issue.
How can we resolve the above issue?

Let me know if you need any more information.

 #### Proposed Solution
As mentioned above for Issue 1.
But don't know the solution for Issue 2.

srickardti, 2022-02-25:
The first issue can be resolved in two ways:

1. set `chip_stack_lock_tracking = "none"` in the project's args.gni file
2. `#define INCLUDE_xSemaphoreGetMutexHolder 1` in the platform's FreeRTOS config: `src/platform/cc13x2_26x2/FreeRTOSConfig.h`

The second issue is that the Matter stack has grown larger than the 704k bytes of flash the device has. For the pump application, this is due to the `is_debug=true` configuration. This configuration sets the optimization for the compiler to `-Og` for debugging. The `is_debug` flag needs to be false to enable the configuration in build/config/compiler/BUILD.gn to optimize for size.

I'll get a PR up to add the mutex holder change for FreeRTOS.

mgarb1, 2022-02-25:
@srickardti I dont think we need https://github.com/project-chip/connectedhomeip/pull/15595 once https://github.com/project-chip/connectedhomeip/pull/15499 is merged.

What is the memory impact enabling: #define INCLUDE_xSemaphoreGetMutexHolder 1 on the platform?

srickardti, 2022-02-25:
It is a 1K flash difference when mutex holder tracking is enabled. I am fine with either fix. But if #15499 is accepted, could you propagate that change to the lock app example.

prashanthumbre, 2022-03-02:
Just an update:

I cloned the latest commit: `169b48a1692f6fb5999c950e4c9e8fa6728166a4 ` and modified the `#define INCLUDE_xSemaphoreGetMutexHolder 1` as suggested.

The `ninja -C out/debug` is successful and files got generated. But after flashing the .out file I got the following log on the CC2652R7's serial terminal:

```
OpenThread ifconfig up and thread start
OpenThread started: OK
Setting OpenThread device type to ROUTER
Initialize Server
Using ZAP configuration...
Chip stack locking error at '../../third_party/connectedhomeip/src/app/util/attribute-storage.cpp:553'. Code is unsafe/racy
chipDie chipDie chipDie
```

closed 2022-03-03 by bzbarsky-apple