project-chip/connectedhomeip #15531: ESP32/M5 board crash looping after flashing newer SHA (2/24) - abort() was called at PC 0x401108cb on core 0

kean-apple, 2022-02-24:
#### Problem
ESP32/M5 board crash looping after flashing newer SHA: e89ad5b04eacf95663c0da53bf71743897174d28

Did not see this issue yesterday with SHA: 9f09e5290e5ce97f446158ac83aaf1942fc4f540

abort() was called at PC 0x401108cb on core 0
0x401108cb: chip::Platform::VerifyInitialized(char const*) at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:477
 (inlined by) VerifyInitialized at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/lib/support/CHIPMem-Malloc.cpp:60
 (inlined by) VerifyInitialized at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/lib/support/CHIPMem-Malloc.cpp:58

Backtrace:0x40081e16:0x3ffbc8900x40092281:0x3ffbc8b0 0x40098916:0x3ffbc8d0 0x401108cb:0x3ffbc940 0x4011091e:0x3ffbc960 0x40129bb9:0x3ffbc980 0x401290b6:0x3ffbc9b0 0x40128989:0x3ffbc9e0 0x40129283:0x3ffbca00 0x401292fc:0x3ffbca50 0x401166ad:0x3ffbca80 0x40116727:0x3ffbcaa0 0x40116ae9:0x3ffbcac0 0x400f26bb:0x3ffbcbd0 0x400d9ad7:0x3ffbcbf0 0x401c675b:0x3ffbcdf0 0x40095485:0x3ffbce10 
0x40081e16: panic_abort at /Users/keanhinlim/tools/esp-idf/components/esp_system/panic.c:402

0x40092281: esp_system_abort at /Users/keanhinlim/tools/esp-idf/components/esp_system/esp_system.c:121

0x40098916: abort at /Users/keanhinlim/tools/esp-idf/components/newlib/abort.c:46

0x401108cb: chip::Platform::VerifyInitialized(char const*) at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:477
 (inlined by) VerifyInitialized at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/lib/support/CHIPMem-Malloc.cpp:60
 (inlined by) VerifyInitialized at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/lib/support/CHIPMem-Malloc.cpp:58

0x4011091e: chip::Platform::MemoryCalloc(unsigned int, unsigned int) at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/lib/support/CHIPMem-Malloc.cpp:96

0x40129bb9: chip::Platform::Impl::PlatformMemoryManagement::MemoryCalloc(unsigned int, unsigned int) at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/lib/support/ScopedBuffer.h:121
 (inlined by) chip::Platform::Impl::ScopedMemoryBufferBase<chip::Platform::Impl::PlatformMemoryManagement>::Calloc(unsigned int, unsigned int) at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/lib/support/ScopedBuffer.h:106
 (inlined by) chip::Platform::ScopedMemoryBuffer<char, chip::Platform::Impl::PlatformMemoryManagement>::Calloc(unsigned int) at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/lib/support/ScopedBuffer.h:152
 (inlined by) chip::DeviceLayer::Internal::ESP32Config::WriteConfigValueStr(chip::DeviceLayer::Internal::ESP32Config::Key, char const*, unsigned int) at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/platform/ESP32/ESP32Config.cpp:273

0x401290b6: chip::DeviceLayer::ConfigurationManagerImpl::WriteConfigValueStr(chip::DeviceLayer::Internal::ESP32Config::Key, char const*, unsigned int) at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/platform/ESP32/ConfigurationManagerImpl.cpp:253

0x40128989: chip::DeviceLayer::Internal::GenericConfigurationManagerImpl<chip::DeviceLayer::Internal::ESP32Config>::StoreUniqueId(char const*, unsigned int) at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericConfigurationManagerImpl.cpp:581

0x40129283: chip::DeviceLayer::Internal::GenericConfigurationManagerImpl<chip::DeviceLayer::Internal::ESP32Config>::Init() at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericConfigurationManagerImpl.cpp:68

0x401292fc: chip::DeviceLayer::ConfigurationManagerImpl::Init() at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/platform/ESP32/ConfigurationManagerImpl.cpp:95

0x401166ad: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_InitChipStack() at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.cpp:77

0x40116727: chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::_InitChipStack() at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.cpp:82

0x40116ae9: chip::DeviceLayer::PlatformManagerImpl::_InitChipStack() at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/platform/ESP32/PlatformManagerImpl.cpp:125

0x400f26bb: chip::DeviceLayer::PlatformManager::InitChipStack() at /Users/keanhinlim/connectedhomeip/src/include/platform/PlatformManager.h:329
 (inlined by) chip::DeviceLayer::PlatformManager::InitChipStack() at /Users/keanhinlim/connectedhomeip/src/include/platform/PlatformManager.h:315
 (inlined by) chip::DeviceManager::CHIPDeviceManager::Init(chip::DeviceManager::CHIPDeviceManagerCallbacks*) at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/../main/CHIPDeviceManager.cpp:64

0x400d9ad7: app_main at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/../main/main.cpp:696

0x401c675b: main_task at /Users/keanhinlim/tools/esp-idf/components/freertos/port/port_common.c:129

0x40095485: vPortTaskWrapper at /Users/keanhinlim/tools/esp-idf/components/freertos/port/xtensa/port.c:130

kean-apple, 2022-02-24:
[M5-crash-looping.txt](https://github.com/project-chip/connectedhomeip/files/8135074/M5-crash-looping.txt)

chshu, 2022-02-25:
Fixed by https://github.com/project-chip/connectedhomeip/pull/15507

closed 2022-02-25 by chshu

kean-apple, 2022-02-25:
Awesome. thanks. its working again on 448578dd09004679d10a7fa883f2e680de05079d