project-chip/connectedhomeip #15550: unable to pair iOS device with M5 ESP32

plauric, 2022-02-24:
#### Problem

I'm running the all-clusters-app/esp32 example on a M5. The BLE pairing with the iOS device fails with this message:

 "E (75968) chip[BLE]: HandleChipConnectionReceived failed, err = 417

iOS version: 15.3.1
Matter repo: 83aac592cedfa04cb2941bb0f75f2a4c44ddf799
esp-idf: v5.0-dev (I also tried v4.4 and v4.3 and had the same issue)

Thank you for any suggestion.

-----------------

I (97) boot:  4 ota_1            OTA app  �ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6708
load:0x40078000,len:14812
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v5.0-dev 2nd stage bootloader
I (27) boot: compile time 11:36:10
I (27) boot: chip revision: 3
I (30) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (82) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (90) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (97) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (105) boot:  5 ot_storage       Unknown data     01 3a 00317000 00002000
I (112) boot: End of partition table
I (117) boot: No factory image, trying OTA 0
I (121) boot_comm: chip revision: 3, min. application chip revision: 0
I (129) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=30e60h (200288) map
I (210) esp_image: segment 1: paddr=00050e88 vaddr=3ffbdb60 size=08508h ( 34056) load
I (223) esp_image: segment 2: paddr=00059398 vaddr=40080000 size=06c80h ( 27776) load
I (235) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=e2bd0h (928720) map
I (571) esp_image: segment 4: paddr=00142bf8 vaddr=40086c80 size=177c4h ( 96196) load
I (611) esp_image: segment 5: paddr=0015a3c4 vaddr=50000000 size=00010h (    16) load
I (627) boot: Loaded app from partition at offset 0x20000
I (672) boot: Set actual ota_seq=1 in otadata[0]
I (672) boot: Disabling RNG early entropy source...
I (683) cpu_start: Pro cpu up.
I (684) cpu_start: Starting app cpu, entry point is 0x400814bc
0x400814bc: call_start_cpu1 at /home/petru/tools/esp-idf/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.
I (699) cpu_start: Pro cpu start user code
I (700) cpu_start: cpu freq: 160000000
I (700) cpu_start: Application information:
I (704) cpu_start: Project name:     chip-all-clusters-app
I (710) cpu_start: App version:      test_event_4_2021_07_06-1968-g8
I (717) cpu_start: Compile time:     Feb 24 2022 17:25:56
I (723) cpu_start: ELF file SHA256:  630f05ab87f3c167...
I (729) cpu_start: ESP-IDF:          v5.0-dev
I (735) heap_init: Initializing. RAM available for dynamic allocation:
I (742) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (748) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
[... 249 lines cut ...]
I (90978) NimBLE: GAP procedure initiated: advertise; 
I (90978) NimBLE: disc_mode=2
I (90978) NimBLE:  adv_channel_map=0 own_addr_type=2 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (90988) NimBLE: 

I (90998) chip[DL]: CHIPoBLE advertising started
I (91008) app-devicecallbacks: Current free heap: 72820

I (91038) chip[DL]: BLE GAP connection terminated (con 0 reason 0x13)
I (91038) app-devicecallbacks: CHIPoBLE disconnected
I (91048) app-devicecallbacks: Current free heap: 72820

I (91058) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable, device name MATTER-3840)
I (91068) chip[DL]: Device already advertising, stop active advertisement and restart
I (91068) NimBLE: GAP procedure initiated: stop advertising.

I (91088) NimBLE: GAP procedure initiated: advertise; 
I (91088) NimBLE: disc_mode=2
I (91088) NimBLE:  adv_channel_map=0 own_addr_type=2 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (91098) NimBLE: 

I (97938) all-clusters-app: Display going to sleep...

 #### Proposed Solution
<suggested fix, suggested enhancement>

dhrishi, 2022-02-25:
Hi @plauric Can you please check if #15299 is related, specifically: https://github.com/project-chip/connectedhomeip/issues/15299#issuecomment-1050574537

plauric, 2022-02-28:
@dhrishi Thank you very much! That was it - the iOS device must run 15.4 developer beta.

closed 2022-02-28 by plauric