project-chip/connectedhomeip #15560: Default address resolver impl may create non-routable PeerAddresses

bzbarsky-apple, 2022-02-25:
#### Problem
In `AddressResolve_DefaultImpl.cpp` the code in `Resolver::OnNodeIdResolved` does:
```
         address.SetInterface(nodeData.mInterfaceId);
```
unconditionally.  But the interface id can be wrong for a non-link-local address: the interface the DNS-SD response comes in on may not match the interface that needs to be used to communicate with the relevant IP.  See comments in `OperationalDeviceProxy::ToPeerAddress` for example.

As a result, using the `PeerAddress` `OnNodeIdResult` creates can lead to inability to deliver messages to the resolved IP.

 #### Proposed Solution
Either implement logic to only set the interface id if `IsIPv6LinkLocal` is true, or very clearly document the limitations of the PeerAddress that this API produces in the API documentation.

@andy31415

bzbarsky-apple, 2022-02-25:
This is code that was added in https://github.com/project-chip/connectedhomeip/pull/15200

andy31415, 2022-02-25:
Why does the interface the mdns packet comes on not support the relevant IP?

this is easy enough to change, however I am very surprised by this. My understanding is that getting rid of the interface just makes it route in case it happens to be routable over other interfaces in case one goes down.

cecille, 2022-02-25:
From the mdns spec s14:
```
   Except in the case of proxying and other similar specialized uses,
   addresses in IPv4 or IPv6 address records in Multicast DNS responses
   MUST be valid for use on the interface on which the response is being
   sent.

   Just as the same link-local IP address may validly be in use
   simultaneously on different links by different hosts, the same link-
   local host name may validly be in use simultaneously on different
   links, and this is not an error.  A multihomed host with connections
   to two different links may be able to communicate with two different
   hosts that are validly using the same name.  While this kind of name
   duplication should be rare, it means that a host that wants to fully
   support this case needs network programming APIs that allow
   applications to specify on what interface to perform a link-local
   Multicast DNS query, and to discover on what interface a Multicast
   DNS response was received.
```
If a device is receiving A/AAAA records on an interface, the addresses specified in those records should be routable on that interface. That interface may not be the PREFERRED route out, but it is routeable. If a network stack does not wish to make use of the interface indicator for non-link-local addresses and allow the network stack to determine the route, that seems like a reasonable optimization, but if you really cannot route an address on the interface on which the record was received, that is a different problem.

bzbarsky-apple, 2022-02-25:
> Why does the interface the mdns packet comes on not support the relevant IP?

I don't know offhand, but @sagar-apple was definitely seeing this with Thread networks.