project-chip/connectedhomeip #15562: Error during Building the ESP32 in the all-clusters-app

rinelb, 2022-02-25:
Building the ESP32 in the all-clusters-app following the instruction in the [README.md](https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/esp32#readme) 

when i run >>idf.py build 
error attached

[error.txt](https://github.com/project-chip/connectedhomeip/files/8139941/error.txt)

dont know what the error is

dhrishi, 2022-02-25:
@rinelb Can you confirm if you have used IDF release tag v4.4 as mentioned in the README.md (and not master)? CHIP supports IDF stable release v4.4 and may not complile with master. A log/print in your error.txt is present in ESP-IDF master branch but not in v4.4:

` #warning driver/periph_ctrl.h header is no longer used, and will be removed in future versions.`

rinelb, 2022-02-28:
@dhrishi yes im using
csir@csir-OptiPlex-790:esp-idf$ idf.py --version
ESP-IDF v4.4

dhrishi, 2022-02-28:
@rinelb The above error should not be seen in your logs then. Can you share the log of all the steps you've executed from README.md. Also, can you provide details about the platform used for setup (Linux/Mac)?

harami-eng, 2022-03-02:
@dhrishi  I am getting an error with the same sample as this issue.
I am not able to set the file reference in my environment.
Could you please tell me about this situation? #15665

rinelb, 2022-03-02:
unfortunately i couldnt get it working using ubuntu 20 lte, i just did a clean install of ubuntu 18 LTE (completely removed ubuntu 20), i followed the same instruction on the web as before, i used python version 3.8 (python 3.8-venv and python 3.8-dev)  

now i flash the code on the ESP32

harami-eng, 2022-03-02:
@rinelb Thanks for the info. It is very helpful. I will try it as soon as possible.

harami-eng, 2022-03-03:
@rinelb I installed a new ubuntu 18.04.6 LTS and tried, but could not compile.

Please let me know the output.
cat /etc/lsb-release

rinelb, 2022-03-03:
@harami-eng 

csir@csir-OptiPlex-790:~$ cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.6 LTS"

**connectedhomeip is the master branch, right?** 
i beleive so after removing ubuntu completely and installing it, below was the steps i used that worked for me

$ mkdir ${HOME}/tools
  $ cd ${HOME}/tools
  $ git clone https://github.com/espressif/esp-idf.git
  $ cd esp-idf
  $ git checkout v4.4
  $ git submodule update --init
  $ ./install.sh
  $ sudo apt-get install ninja-build
$ cd ${HOME}/tools/esp-idf
$ ./install.sh
$ . ./export.sh

then i went to home directory 
>> cd ~

>>git clone https://github.com/project-chip/connectedhomeip.git
>>sudo apt-get install git gcc g++ python pkg-config libssl-dev libdbus-1-dev \
     libglib2.0-dev libavahi-client-dev ninja-build python3-venv python3-dev \
     python3-pip unzip libgirepository1.0-dev libcairo2-dev
>> sudo apt-get install python3.8-venv
>>sudo apt-get install python3.8-dev
then switch my python3 default version from 3.6 to 3.8
>>sudo update-alternatives --install /usr/bin/python3 python3 /usr/bin/python3.6 1
>>sudo update-alternatives --install /usr/bin/python3 python3 /usr/bin/python3.8 2

then check if your python version changed to 3.8 if it didnt then run and select python 3.8
>>sudo update-alternatives --config python3

>>cd ~/connectedhomeip
>>source scripts/bootstrap.sh

then i went to 
$  cd ~/connectedhomeip/examples/all-clusters-app/esp32/
$ idf.py set-target esp32
$ idf.py build

the build went successfully 

csir@csir-OptiPlex-790:~/connectedhomeip$ git branch 
* master

csir@csir-OptiPlex-790:~/connectedhomeip$ python3 --version
Python 3.8.12

csir@csir-OptiPlex-790:~/connectedhomeip$ cmake --version
cmake version 3.10.2
CMake suite maintained and supported by Kitware (kitware.com/cmake).

been almost two months trying to build it  ... hope above works for u

harami-eng, 2022-03-03:
@rinelb Thank you.
I've compiled it and now I'm going to check the operation.
I wish the README.md was as detailed as this.
My trials are going to be as long as yours... 
Anyway, thanks for the progress!

dhrishi, 2022-03-03:
@rinelb Thanks for the detailed steps. Please close the issue if compilation is successful.
@harami-eng Good to know that you got it working too. Please close #15665

closed 2022-03-03 by rinelb