project-chip/connectedhomeip #15573: Use-After-Free bug when closing EndPoints

andyp-apple, 2022-02-25:
The `EndPoint` objects (`UDPEndPoint` and `TCPEndPoint`) are created by `EndpointManager<EndPointType>::NewEndPoint()`, which will return the `Endpoint` pointer while also saving it in an `ObjectPool` (`EndPointManagerImplPool::sEndPointPool`).
`EndPoint` objects are reference counted (class derived from `ReferenceCounted`): they are freed only when the reference counter decrements to zero. When the reference counter reaches 0 the endpoint will also be removed from the `ObjectPool`.

However by sending a signal such as SIGINT or SIGTERM, the `ObjectPool` will be removed before the `EndPoint` is released, which ends up calling `chip::HeapObjectPool<chip::Inet::UDPEndPointImplSockets>::ReleaseObject(chip::Inet::UDPEndPointImplSockets *)`, which deletes the object without using the reference counter.

https://github.com/project-chip/connectedhomeip/blob/980e1770075e5209d719693bb2e70d1966e4949f/src/lib/support/Pool.h#L340-L353

`Platform::Delete(object)` calls the object destructor and frees its memory. Since there is still a dangling pointer that references that object, it leads to a Use-After-Free situation: any further use of the object will use invalid memory that an attacker could potentially re-allocate.

`chip::HeapObjectPool<T>::ReleaseObject(T *)` doesn't seem to consider the possibility that the object stored in the `HeapObjectPool` may be a sub-class of `ReferenceCounted`.

A solution might be to have a different version of that method which calls `objectRelease()` instead of `Platform::Delete(object)` if the template T is a sub-class of `ReferenceCounted`

bzbarsky-apple, 2022-08-06:
@andyp-apple It doesn't matter that these are ReferenceCounted; the same issue would happen with a non-refcounted thing: whoever has not released it back to the pool yet is holding a dangling pointer if the pool has gone away, right?

The point is you're not supposed to destroy pools while you are using objects from them, if you plan to keep using those objects.

A few other notes:
1. The behavior here is different depending on whether ASan is being used.  If ASan is _not_ used, then `~HeapObjectPool` will `VerifyOrDie(Allocated() == 0)`.  If ASan is used, then it will try to free all the objects, so ASan can catch use-after-free if they're used after the pool is destroyed.
2. This specific pool is a static variable, which is why its destructor ends up running on SIGINT or SIGTERM: the process is shutting down.  I would _expect_ that after the start of running static destructors we no longer touch the endpoints, but if you are seeing otherwise we should definitely look into that.

bzbarsky-apple, 2022-08-10:
Some more details:

* The original report here was about `chip-shell` on Linux.  The issue was seen when compiling it with asan (makes sense), then doing something with it do make sure it creates endpoints (what exactly is unclear) then hitting Ctrl-C.
* It's possible that this has already been fixed; I seem to recall us having shutdown issues with endpoint pools that were fixed....

jtung-apple, 2022-08-23:
Had a quick sync-up with @andyp-apple, who mentioned that this issue is not reproducing any more. Closing.

closed 2022-08-23 by jtung-apple