project-chip/connectedhomeip #15584: Properly integrate LastKnownGood time processing for CASE cert expiries

tcarmelveilleux, 2022-02-25:
#### Problem
Spec has clear mention of "Last Known Good time" to be used for certificate expiries. If no last known good time is available, CASE can become impossible (see #14347), unless we allow the spec-allowed policy of overriding expiry checks. The SDK does not currently handle "Last known good time", or allow any method of overriding expiry, even though the spec says:

> 3.5.6. Time and date considerations for certificate path validation
>
> The Basic Path Validation algorithm in [RFC 5280] mandates the consideration of the "current time" against the validity period (notBefore, notAfter fields) when validating paths. The usage of "current time" assumes that such a time is available and correct, which is a strong assumption when considering some constrained devices or devices only locally reachable on a network in the absence of infrastructure to synchronize time against a global real-time reference.
> 
> When the Crypto_VerifyChain primitive is used, rather than overriding the Basic Path Validation algorithm of [RFC 5280] Nodes SHALL consider the following definition of "current time" that accounts for the possible lack of a real time reference:
> 
> If a Node has a current real-time clock value which is trusted according to implementation-defined means to be accurate with regard to global real-time, whether using [Time Synchronization] features of this specification or other means, then it SHALL use that time;
> 
> Otherwise, the current time SHALL be set to the last-known-good UTC time.
> 
> Upon failure to validate a certificate path, where the only reason for failure is an invalid validity period of a path element, a Node MAY apply a policy of its choice to determine whether to ignore this failure and still consider the path valid.
> 

#### Proposed Solution
- Implement proper minimal handling of last known-good-time, including allowing for override of expiry policy for CASE

Damian-Nordic, 2022-04-04:
Does this involve support for `SetUtcTime` command of the Time synchronization cluster (which I thought would be out of 1.0 scope)? Or how the last known good UTC will be updated?

msandstedt, 2022-06-07:
Fixed with  #19119

closed 2022-06-07 by msandstedt

msandstedt, 2022-06-07:
> Does this involve support for `SetUtcTime` command of the Time synchronization cluster (which I thought would be out of 1.0 scope)? Or how the last known good UTC will be updated?

@Damian-Nordic ,

No, it doesn't.  See #19119.  Last known good time is required to be updated with 1) firmware build time and thereafter 2) NotBerfore times from incoming certificates installed at commissioning time.

Additionally, the spec encourages that nodes update Last Known Good Time with other / better time sources if these are available.  However, with time sync mostly de-scoped from 1.0, other / better time sources may not be available for nodes that don't provide real time / wall clock time through platform-specific means.

Given this, #19119 adopts the minimal spec-compliant and most portable strategy of only updating Last Known Good Time from firmware build time and incoming certificate NotBefore timestamps.  However, a Setter is provided to update Last Known Good Time on demand:

https://github.com/project-chip/connectedhomeip/blob/8cc291d723116d8fba0a6754c32170d0eb1320a3/src/credentials/FabricTable.h#L443

This is what platforms would call if they happen to have a better / newer time source.

With the Last Known Good Time feature, a minimal type of time keeping is available on all nodes so long as the given platform can provide firmware build time (all in-tree examples do).  If it is desired to enact a certificate validity policy based on the Last Known Good Time, this can be enabled by injecting a Validity policy like the following: https://github.com/project-chip/connectedhomeip/blob/master/src/credentials/examples/LastKnownGoodTimeCertificateValidityPolicyExample.h

By default, [CHIPCert.cpp](https://github.com/project-chip/connectedhomeip/blob/8cc291d723116d8fba0a6754c32170d0eb1320a3/src/credentials/CHIPCert.cpp#L437) does not enable validity checks based on Last Known Good Time.  This decision was made in consideration of the fact that Last Known Good Time can be corrupted if commissioners inject certificates with nonsensical (future) NotBefore times.  Given that, validity checks based upon Last Known Good Time are an opt-in feature.  Opt-in is achieved by injecting a validity policy with such enforcement.

Both types of behavior (opt in or opt out) are permissible under the spec:

> Upon failure to validate a certificate path, where the only reason for failure is an invalid validity
period of a path element, a Node MAY apply a policy of its choice to determine whether to ignore
this failure and still consider the path valid.