project-chip/connectedhomeip #15588: esp32 example failed to resolve component

XingpingGY, 2022-02-25:
#### Problem
<what's wrong or missing, please include any applicable:
 * I just tried to build the formal examples in examples/all-clusters-app/esp32 project
 * `CMake Error at C:/Users/911/tools/Espressif/frameworks/esp-idf-v4.4/tools/cmake/build.cmake:200 (message):
  Failed to resolve component 'tft'.
Call Stack (most recent call first):
  C:/Users/911/tools/Espressif/frameworks/esp-idf-v4.4/tools/cmake/build.cmake:231 (__build_resolve_and_add_req)
  C:/Users/911/tools/Espressif/frameworks/esp-idf-v4.4/tools/cmake/build.cmake:512 (__build_expand_requirements)
  C:/Users/911/tools/Espressif/frameworks/esp-idf-v4.4/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:34 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/911/connectedhomeip/examples/all-clusters-app/esp32/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1`

-- Configuring incomplete, errors occurred!
See also "C:/Users/911/connectedhomeip/examples/all-clusters-app/esp32/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >

 #### Proposed Solution
<suggested fix, suggested enhancement>

dhrishi, 2022-02-25:
@XingpingGY Can you confirm if you have updated the chip submodules using `git submodule update --init --recursive`? As component tft is at `examples/common/m5stack-tft/repo/components/tft/` which is a part of `m5stack-tft/repo` submodule.

XingpingGY, 2022-02-26:
Thank you for your comments. I will check it carefully.

dhrishi, 2022-03-03:
@XingpingGY Please confirm if this works fine now and close the issue if it does.

dhrishi, 2022-03-08:
Closing due to inactivity. Feel free to re-open

closed 2022-03-08 by dhrishi

aaaaa654, 2023-03-11:
> @XingpingGY Can you confirm if you have updated the chip submodules using `git submodule update --init --recursive`? As component tft is at `examples/common/m5stack-tft/repo/components/tft/` which is a part of `m5stack-tft/repo` submodule.

Hey , i am having the same issue, i tried running the above command in the main 'connectedhomeip' directory and it installed a bunch of things but the issue is still there? What else should i try?

oidebrett, 2023-05-05:
> Hey , i am having the same issue, i tried running the above command in the main 'connectedhomeip' directory and it installed a bunch of things but the issue is still there? What else should i try?

I had the same problem and after some investigation I think that the m5stack-tft repo submodules are not being brought in. Even when you try `git submodule update --init --recursive`

My solution was to run this command at the connectedhomeip root folder.
./scripts/checkout_submodules.py --recursive --force --platform esp32