project-chip/connectedhomeip #15599: Failed to restart OTA Transfer after aborting midway

ukatira, 2022-02-25:
#### Problem
Steps
1. Initiate OTA Transfer using Linux apps.
2. Kill Provider App to abort transfer midway.
3. Relaunch Provider App and initiate another Query to restart OTA Transfer.

Seeing error messages on Provider side when sending new Query
[1645751391688] [1109:20414715] CHIP: [IN] Data received on an unknown session (LSID=1). Dropping it!
[1645751396689] [1109:20414715] CHIP: [IN] Data received on an unknown session (LSID=1). Dropping it!
[1645751398689] [1109:20414680] CHIP: [IN] Data received on an unknown session (LSID=1). Dropping it!
[1645751400689] [1109:20414680] CHIP: [IN] Data received on an unknown session (LSID=1). Dropping it!

 #### Proposed Solution
Restarting OTA Transfer should work.

[ota-p.txt](https://github.com/project-chip/connectedhomeip/files/8143445/ota-p.txt)
[ota-r.txt](https://github.com/project-chip/connectedhomeip/files/8143446/ota-r.txt)

isiu-apple, 2022-03-05:
This issue tracks the resume portion of the issue mentioned in https://github.com/project-chip/connectedhomeip/issues/11760

Section 11.20.3.5 Pg 666: The ReceiveInit message from the OTA Requestor MAY have the RC[STARTOFS] bit set and associated STARTOFS field set to indicate the resumption of a transfer previously aborted, or to affect partial windowed access to the portion of a Software Image desired.

ukatira, 2022-03-07:
@isiu-apple This issue is to track restarting the OTA Transfer when aborting midway which should work with V1.0
We can track the resume part here: https://github.com/project-chip/connectedhomeip/issues/15925

closed 2022-03-09 by carol-apple