project-chip/connectedhomeip #15612: Make PairingDelegate callbacks clearer

bzbarsky-apple, 2022-02-25:
#### Problem
PairingDelegate is a mishmash of "notify about PASE established", "notify about CASE established" (using the same callback!), etc. 

When you get OnPairingComplete you have no idea what it actually means, especially if you are trying to commission one device while establishing CASE to another unrelated device.

 #### Proposed Solution
Come up with a clear definition of what the callbacks on this interface mean (changing the interface as needed in the process), implement them.

cecille, 2022-03-03:
related - https://github.com/project-chip/connectedhomeip/issues/15409. I think that issue is s result of different assumptions about what the pairing callback means.

caipiblack, 2022-03-15:
Currently If I do the commissioning of a thread device from the application, we have to see the Matter logs to know what happens, 

I think it should be good to implement more states in the `OnStatusUpdate()` callback so we can use it to display "commissioning status" on the final application.

For example:
- Searching the device (BLE, SoftAP, OnNetwork)
- Device discovered (BLE, SoftAP, OnNetwork)
- Waiting for the device in the end network (should be called when we know the device is going to connect on final network)
- Device discovered in the end network

Or better: add a status for each commissioning steps in `chip::Controller::DevicePairingDelegate::Status` ! 

```
    enum Status : uint8_t
    {
        SecurePairingSuccess = 0,
        SecurePairingFailed,
    };

    /////////// callbacks from DevicePairingDelegate /////////
    void OnStatusUpdate(chip::Controller::DevicePairingDelegate::Status status) override;
    void OnPairingComplete(CHIP_ERROR error) override;
    void OnCommissioningComplete(NodeId deviceId, CHIP_ERROR error) override;
```

tehampson, 2023-02-09:
Just to add more context to this issue so that when someone is working on a solution they get a slightly bigger picture of that was spotted while working on https://github.com/project-chip/connectedhomeip/pull/24919.

Currently when `SetUpCodePairer::OnDeviceDiscoveredTimeoutCallback` it will call `OnSessionEstablishmentError` which ends up calling `DeviceCommissioner::RendezvousCleanup`. In the case where discovery times out, because we have no PASE connection being established this particular case is not calling `OnPairingComplete`. Because they are not getting `OnPairingComplete` callback they have hooked into `OnStatusUpdate` where `status == Status::SecurePairingFailed` where they are not getting the proper error code. When considering a solution to make PairingDelegate callbacks clearer, consider the case where discovery fails and what kind of call back would be more clear