project-chip/connectedhomeip #15615: ESP32/M5 IDF menuconfig missing option for setup pin code and discriminator

kean-apple, 2022-02-25:
#### Problem

SHA: 448578dd09004679d10a7fa883f2e680de05079d
1. Run idf.py menuconfig
2. Go to Component Config -> CHIP Device Layer-> Testing Options
 - missing option to set setup pin code & decriminator

This use to be present a while back. Without this option it makes it hard to pair multiple M5 boards on the same network

[image]

dhrishi, 2022-02-26:
@kean-apple I think that it was removed in https://github.com/project-chip/connectedhomeip/pull/14676 on purpose.

Discriminator is taken from persistent storage of the platform and if not available then the test value is used from [here](https://github.com/project-chip/connectedhomeip/blob/master/src/include/platform/CHIPDeviceConfig.h#L900). Similarly, setup pin code is calculated from the spake2 verifier and if not available then the test value is used from [here](https://github.com/project-chip/connectedhomeip/blob/master/src/include/platform/CHIPDeviceConfig.h#L890)

dhrishi, 2022-03-03:
Closing this as it is addressed.

closed 2022-03-03 by dhrishi