project-chip/connectedhomeip #15620: New Testscripts are not running

kowsisoundhar12, 2022-02-28:
Summary :
I have added 4 Test scripts from Secure Channel,after the test Execution noticed added testscripts are not executed 

Execution screenshot and logs:
[image]
[image]
[Execution_log (1).zip](https://github.com/project-chip/connectedhomeip/files/8150333/Execution_log.1.zip)

Note :
1.i have tested this old commit id "593b427fffc850cade435523beed340f741fb6d7",in this commit id we can able to add & test new scripts ,this particular commit id date is feb 22
2.i have tested one more commit id "09af03fb005f0a4fe6b3ba4741dc43eebbe56bb6",in this commit id newly added  testscripts are not supporting ,this particular commit id date is feb23

raju-apple, 2022-03-02:
@kowsisoundhar12 @manjunathambiger23 Can you guys try with this command ? `export NODE_OPTIONS="--max-old-space-size=5120"`and see if it still fails. Spoke to Vivien briefly and he mentioned that he has seen the nodes runner memory going out of space for chip-tool

raju-apple, 2022-03-02:
A quick way to confirm is to just run : `./scripts/tools/zap_regen_all.py --type tests --tests chip-tool`

kowsisoundhar12, 2022-03-03:
@raju-apple I have ran this command "./scripts/tools/zap_regen_all.py --type tests --tests chip-tool",EXecution got failed
[image]

closed 2022-05-10 by kowsisoundhar12