project-chip/connectedhomeip #15621: [esp32] failed to resolve component 'hip'

XingpingGY, 2022-02-28:
I followed the markdown file in `example/all-cluster-app/esp32` step by step but failed.
And I cannot solve this problem.

`C:\Users\HIU7SGH\tools\connectedhomeip\examples\all-clusters-app\esp32>idf.py menuconfig
Executing action: menuconfig
Running cmake in directory c:\users\hiu7sgh\tools\connectedhomeip\examples\all-clusters-app\esp32\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 c:\users\hiu7sgh\tools\connectedhomeip\examples\all-clusters-app\esp32"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: C:/Users/HIU7SGH/tools/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1")
-- Component directory C:/Users/HIU7SGH/tools/Espressif/frameworks/esp-idf-v4.4/examples/common_components/cmake does not contain a CMakeLists.txt file. No component will be added
-- Component directory C:/Users/HIU7SGH/tools/Espressif/frameworks/esp-idf-v4.4/examples/common_components/credentials does not contain a CMakeLists.txt file. No component will be added
-- Component directory C:/Users/HIU7SGH/tools/Espressif/frameworks/esp-idf-v4.4/examples/common_components/m5stack-tft does not contain a CMakeLists.txt file. No component will be added
-- Component directory C:/Users/HIU7SGH/tools/Espressif/frameworks/esp-idf-v4.4/examples/common_components/pigweed does not contain a CMakeLists.txt file. No component will be added
-- Component directory C:/Users/HIU7SGH/tools/Espressif/frameworks/esp-idf-v4.4/examples/common_components/tracing does not contain a CMakeLists.txt file. No component will be added
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/HIU7SGH/tools/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/HIU7SGH/tools/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/HIU7SGH/tools/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at C:/Users/HIU7SGH/tools/Espressif/frameworks/esp-idf-v4.4/tools/cmake/build.cmake:200 (message):
  Failed to resolve component 'chip'.
Call Stack (most recent call first):
  C:/Users/HIU7SGH/tools/Espressif/frameworks/esp-idf-v4.4/tools/cmake/build.cmake:231 (__build_resolve_and_add_req)
  C:/Users/HIU7SGH/tools/Espressif/frameworks/esp-idf-v4.4/tools/cmake/build.cmake:512 (__build_expand_requirements)
  C:/Users/HIU7SGH/tools/Espressif/frameworks/esp-idf-v4.4/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:34 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/HIU7SGH/tools/connectedhomeip/examples/all-clusters-app/esp32/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1`

 #### Proposed Solution
<I have used `git sbumodule --init` but it is of no use>

bzbarsky-apple, 2022-05-21:
@dhrishi

SulianEffInnov, 2022-05-31:
Hello, I'm encountering the same issue.
Does anyone have an idea to deal with it ?

bzbarsky-apple, 2023-01-13:
@XingpingGY Please see question above from @shubhamdp?

closed 2023-10-15 by stale[bot]