project-chip/connectedhomeip #15626: [TC-TSTAT-2.1]-In Thermostat Cluster few attributes needs to developed in chip-tool.

Rajashreekalmane, 2022-02-28:
#### Problem
CHIP-tool SDK software has to implement the the following attributes mentioned in Thermostat Cluster.

1. OccupiedSetbackMin attribute
2. OccupiedSetbackMax attribute
3. UnoccupiedSetback attribute
4. UnoccupiedSetbackMin attribute
5. UnoccupiedSetbackMax attribute 
6. EmergencyHeatDelta attribute

Reference Screenshot:
[image]

Spec Reference:
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/Thermostat.adoc#ref_sec_6_3_2_2_4_4

andy31415, 2022-05-03:
These seem like optional attributes

Rajashreekalmane, 2022-06-01:
@andy31415 All are mandatory attributes Expect EmergencyHeatDelta attribute based on the feature these attributes are supporting.

bzbarsky-apple, 2022-06-13:
These are all supported and have been for a while.

closed 2022-06-13 by bzbarsky-apple