project-chip/connectedhomeip #15631: DataVersion handling for dynamic endpoint 0

chiragatal, 2022-02-28:
#### Problem
* actual behavior:
 In `Server::Init()` the data model is initialized `InitDataModelHandler()` and then the fabric table is initialized `mFabrics.Init()`. In the initialisation of the fabrics, `IncreaseClusterDataVersion()` is called indirectly for endpoint_id `0`, and cluster_id `OperationalCredentials::Id`.
Now if endpoint 0 is created dynamically using `emberAfSetDynamicEndpoint()`, the endpoint is not yet created and so the DataVersion is not found, and this prints an error `Endpoint 0, Cluster 0x0000_003E not found in IncreaseClusterDataVersion!`. The dynamic endpoints cannot be created before `Server::Init()`, since the data model itself is being initialized in `Server::Init()`.

bzbarsky-apple, 2022-02-28:
Apart from printing the error message this should be working fine, right?

chiragatal, 2022-03-07:
Yes, I did not see any other issues.

bzbarsky-apple, 2022-03-18:
The relevant stack seems to be:
```
  * frame #0: 0x0000000100021a9c chip-all-clusters-app`chip::app::(anonymous namespace)::IncreaseClusterDataVersion(aConcreteClusterPath=0x00007ff7bfeff0a8) at ember-compatibility-functions.cpp:269:55
    frame #1: 0x0000000100021a3e chip-all-clusters-app`MatterReportingAttributeChangeCallback(endpoint=0, clusterId=62, attributeId=1) at ember-compatibility-functions.cpp:1002:5
    frame #2: 0x000000010006bd88 chip-all-clusters-app`fabricListChanged() at operational-credentials-server.cpp:276:5
    frame #3: 0x00000001000733f7 chip-all-clusters-app`OpCredsFabricTableDelegate::OnFabricRetrievedFromStorage(this=0x0000000100210aa0, fabric=0x0000000100215d58) at operational-credentials-server.cpp:319:9
    frame #4: 0x000000010019b474 chip-all-clusters-app`chip::FabricTable::LoadFromStorage(this=0x0000000100215d58, fabric=0x0000000100215d58) at FabricTable.cpp:628:19
    frame #5: 0x000000010019c44f chip-all-clusters-app`chip::FabricTable::Init(this=0x0000000100215d58, storage=0x0000000100217020) at FabricTable.cpp:774:13
    frame #6: 0x00000001001d15ff chip-all-clusters-app`chip::Server::Init(this=0x0000000100213a78, delegate=0x0000000000000000, secureServicePort=5540, unsecureServicePort=5550, interfaceId=<unavailable>) at Server.cpp:136:20
```

We should think about changing init so we load the fabric table before initializing the data model, which would address this.  But failing that, for now it sounds like it's just a slightly confusing log....