project-chip/connectedhomeip #15639: Pressure measurement [TC-PRS-2.3] - Verification step results are not matching the expected outcome

somu1710, 2022-02-28:
In pressure measurement cluster test case  [TC-PRS-2.3]  is got failed during the execution.
 * expected behavior - Pass , status(0x0)
 * actual behavior - Response Failure: IM Error 0x00000586: General error: 0x86
 * Actual steps -  ./chip-tool pressuremeasurement read tolerance 1 1
                             ./chip-tool pressuremeasurement read scaled-value 1 1
                            ./chip-tool pressuremeasurement read scale 1 1
[image]

andy31415, 2022-04-13:
Pressure management spec marks all of ScaledValue,MinScaledValue,MaxScaledValue,ScaledTolerance and Scale as `EXT`  feature wise.

So if the DUT does not have this feature (feature itself is marked as optional) then it seems reasonable for the device to say 'unsupported attribute'

andy31415, 2022-04-13:
Test case should likely start with 'is feature supported' and only proceed if feature is supported.

The part that is unclear to me is what if we do not build this in our example (which seems spec-correct) yet at a later time some device comes in to certification with this feature and asks for certification. What do we do then?

somu1710, 2022-04-19:
Then in the case optional attributes no need to verify? we have to verify the optional attributes also because other than this some clusters verifying optional attribute also

bzbarsky-apple, 2022-05-21:
You can't require optional attributes to be implemented.  What exactly are you trying to verify?

@cjandhyala

somu1710, 2022-05-27:
Optional attributes are 'not verifiable' for V1.0 as per discussion.So I close the ticket.

closed 2022-05-27 by somu1710