project-chip/connectedhomeip #15656: [ota-provider-app] Add counters for various QueryImageResponse and ApplyImageResponse commands

ukatira, 2022-02-28:
#### Problem
There is no way right now to simulate real world scenarios where Provider replies Busy, gets the image and then Available or AwaitNextAction and then Proceed

 #### Proposed Solution
Add counters for various QueryImageResponse and ApplyImageResponse commands to simulate sending Busy/NotAvailable for a specified number of times and then sending Available or sending AwaitNextAction for a specified number of times and then sending Proceed.

carol-apple, 2022-03-09:
https://github.com/project-chip/connectedhomeip/pull/15939 provides a way for ota-provider-app to be terminated and started back up with a different query image response.

carol-apple, 2022-03-09:
Moving this back into v1. Had an offline discussion with @ukatira and the following test could be done more conveniently with this issue addressed:

> Requestor sends Query, Provider replies Busy with delay of say 180 sec, Requestor will query again based on delay and Provider responds with Available this time and transfer goes through.

closed 2022-03-14 by carol-apple