project-chip/connectedhomeip #15673: AddressSanitizer: heap-use-after-free in mdns::Minimal::TxtResourceRecord::WriteData

vivien-apple, 2022-03-01:
#### Problem

While trying  to get #15106 in CI I end up in a `head-use-after-free` error with a `Linux` local build of the `all-clusters-app` with `is_asan=true`

The error happens when a `AddNOC` command  is received on the server side. The code is trying to call `app::DnssdServer::Instance().AdvertiseOperational()` onto the chip main thread, while the original dnssd configuration has been started on the app main thread.

Here are the stack I gathered:

```
Thread 1 "chip-all-cluste" hit Breakpoint 1, mdns::Minimal::TxtResourceRecord::WriteData (this=0x555557ad1a13 <mdns::Minimal::RecordWriter::Put16(unsigned short)+79>, out=...) at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/lib/dnssd/minimal_mdns/records/Txt.h:55
55          bool WriteData(RecordWriter & out) const override
(gdb) bt
#0  mdns::Minimal::TxtResourceRecord::WriteData(mdns::Minimal::RecordWriter&) const (this=0x555557ad1a13 <mdns::Minimal::RecordWriter::Put16(unsigned short)+79>, out=...) at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/lib/dnssd/minimal_mdns/records/Txt.h:55
#1  0x0000555557b81d9a in mdns::Minimal::ResourceRecord::Append(mdns::Minimal::HeaderRef&, mdns::Minimal::ResourceType, mdns::Minimal::RecordWriter&) const (this=0x6080001629c0, hdr=..., asType=mdns::Minimal::ResourceType::kAdditional, out=...)
    at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/lib/dnssd/minimal_mdns/records/ResourceRecord.cpp:47
#2  0x0000555557b7955e in mdns::Minimal::ResponseBuilder::AddRecord(mdns::Minimal::ResourceType, mdns::Minimal::ResourceRecord const&) (this=0x55555a569260 <chip::Dnssd::(anonymous namespace)::gAdvertiser+5472>, type=mdns::Minimal::ResourceType::kAdditional, record=...)
    at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/lib/dnssd/minimal_mdns/ResponseBuilder.h:91
#3  0x0000555557b7db78 in mdns::Minimal::ResponseSender::AddResponse(mdns::Minimal::ResourceRecord const&) (this=0x55555a569218 <chip::Dnssd::(anonymous namespace)::gAdvertiser+5400>, record=...)
    at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/lib/dnssd/minimal_mdns/ResponseSender.cpp:216
#4  0x0000555557b47145 in mdns::Minimal::TxtResponder::AddAllResponses(chip::Inet::IPPacketInfo const*, mdns::Minimal::ResponderDelegate*) (this=0x6080001629a0, source=0x7fffffff9ac0, delegate=0x55555a569218 <chip::Dnssd::(anonymous namespace)::gAdvertiser+5400>)
    at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/lib/dnssd/minimal_mdns/responders/Txt.h:33
#5  0x0000555557b7c4aa in mdns::Minimal::ResponseSender::Respond(unsigned int, mdns::Minimal::QueryData const&, chip::Inet::IPPacketInfo const*) (this=0x55555a569218 <chip::Dnssd::(anonymous namespace)::gAdvertiser+5400>, messageId=0, query=..., querySource=0x7fffffff9ac0)
    at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/lib/dnssd/minimal_mdns/ResponseSender.cpp:148
#6  0x0000555557b4e94b in chip::Dnssd::(anonymous namespace)::AdvertiserMinMdns::AdvertiseRecords() (this=0x55555a567d00 <chip::Dnssd::(anonymous namespace)::gAdvertiser>) at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/lib/dnssd/Advertiser_ImplMinimalMdns.cpp:797
#7  0x0000555557b49f12 in chip::Dnssd::(anonymous namespace)::AdvertiserMinMdns::Advertise(chip::Dnssd::OperationalAdvertisingParameters const&) (this=0x55555a567d00 <chip::Dnssd::(anonymous namespace)::gAdvertiser>, params=...)
    at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/lib/dnssd/Advertiser_ImplMinimalMdns.cpp:421
#8  0x0000555557b21cec in chip::app::DnssdServer::AdvertiseOperational() (this=0x55555909aec0 <chip::app::DnssdServer::Instance()::instance>) at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/app/server/Dnssd.cpp:281
#9  0x00005555577607b4 in emberAfOperationalCredentialsClusterAddNOCCallback(chip::app::CommandHandler*, chip::app::ConcreteCommandPath const&, chip::app::Clusters::OperationalCredentials::Commands::AddNOC::DecodableType const&) (commandObj=0x61500000da00, commandPath=..., commandData=...)
    at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/app/clusters/operational-credentials-server/operational-credentials-server.cpp:557
#10 0x00005555577d30ae in chip::app::Clusters::OperationalCredentials::DispatchServerCommand(chip::app::CommandHandler*, chip::app::ConcreteCommandPath const&, chip::TLV::TLVReader&) (apCommandObj=0x61500000da00, aCommandPath=..., aDataTlv=...)
    at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/zzz_generated/all-clusters-app/zap-generated/IMClusterCommandHandler.cpp:1208
#11 0x00005555577dc382 in chip::app::DispatchSingleClusterCommand(chip::app::ConcreteCommandPath const&, chip::TLV::TLVReader&, chip::app::CommandHandler*) (aCommandPath=..., aReader=..., apCommandObj=0x61500000da00)
    at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/zzz_generated/all-clusters-app/zap-generated/IMClusterCommandHandler.cpp:1856
#12 0x0000555557818942 in chip::app::InteractionModelEngine::DispatchCommand(chip::app::CommandHandler&, chip::app::ConcreteCommandPath const&, chip::TLV::TLVReader&) (this=0x55555a55e020 <chip::app::sInteractionModelEngine>, apCommandObj=..., aCommandPath=..., apPayload=...)
    at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/app/InteractionModelEngine.cpp:595
#13 0x00005555577f4222 in chip::app::CommandHandler::ProcessCommandDataIB(chip::app::CommandDataIB::Parser&) (this=0x61500000da00, aCommandElement=...) at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/app/CommandHandler.cpp:309
#14 0x00005555577f1665 in chip::app::CommandHandler::ProcessInvokeRequest(chip::System::PacketBufferHandle&&, bool) (this=0x61500000da00, payload=..., isTimedInvoke=false) at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/app/CommandHandler.cpp:140
#15 0x00005555577f00e5 in chip::app::CommandHandler::OnInvokeCommandRequest(chip::Messaging::ExchangeContext*, chip::PayloadHeader const&, chip::System::PacketBufferHandle&&, bool) (this=0x61500000da00, ec=0x55555a565300 <chip::Server::sServer+6368>, payloadHeader=..., payload=..., isTimedInvoke=false)
    at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/app/CommandHandler.cpp:83
#16 0x00005555578153a7 in chip::app::InteractionModelEngine::OnInvokeCommandRequest(chip::Messaging::ExchangeContext*, chip::PayloadHeader const&, chip::System::PacketBufferHandle&&, bool, chip::Protocols::InteractionModel::Status&)
    (this=0x55555a55e020 <chip::app::sInteractionModelEngine>, apExchangeContext=0x55555a565300 <chip::Server::sServer+6368>, aPayloadHeader=..., aPayload=..., aIsTimedInvoke=false, aStatus=@0x7fffffffc2d0: chip::Protocols::InteractionModel::Status::Failure)
    at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/app/InteractionModelEngine.cpp:235
#17 0x0000555557817588 in chip::app::InteractionModelEngine::OnMessageReceived(chip::Messaging::ExchangeContext*, chip::PayloadHeader const&, chip::System::PacketBufferHandle&&)
    (this=0x55555a55e020 <chip::app::sInteractionModelEngine>, apExchangeContext=0x55555a565300 <chip::Server::sServer+6368>, aPayloadHeader=..., aPayload=...) at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/app/InteractionModelEngine.cpp:418
#18 0x0000555557a50243 in chip::Messaging::ExchangeContext::HandleMessage(unsigned int, chip::PayloadHeader const&, chip::Transport::PeerAddress const&, chip::BitFlags<chip::Messaging::MessageFlagValues, unsigned int>, chip::System::PacketBufferHandle&&)
    (this=0x55555a565300 <chip::Server::sServer+6368>, messageCounter=15975659, payloadHeader=..., peerAddress=..., msgFlags=..., msgBuf=...) at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/messaging/ExchangeContext.cpp:483
#19 0x0000555557a55d77 in chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SessionHandle const&, chip::Transport::PeerAddress const&, chip::SessionMessageDelegate::DuplicateMessage, chip::System::PacketBufferHandle&&)
    (this=0x55555a564e70 <chip::Server::sServer+5200>, packetHeader=..., payloadHeader=..., session=..., source=..., isDuplicate=chip::SessionMessageDelegate::DuplicateMessage::No, msgBuf=...) at ../../../examples [... cut ...]
    #29 0x555557a66d9e in chip::SessionManager::OnMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/transport/SessionManager.cpp:417

SUMMARY: AddressSanitizer: heap-use-after-free (/workspaces/connectedhomeip-linux/out/debug/standalone/chip-all-clusters-app+0x25f2e7a) in mdns::Minimal::TxtResourceRecord::WriteData(mdns::Minimal::RecordWriter&) const
```

 #### Proposed Solution

Either makes the minimal implementation thread-safe or at least ensure that everything happens on the same thread.

vivien-apple, 2022-03-07:
Steps to reproduce on Darwin.
Build with `minimal` instead of `platform` in https://github.com/project-chip/connectedhomeip/blob/b3c3799d8f3ec7e3ab3fb0aea44d5ad55128a354/src/platform/device.gni#L78 and with `is_asan=true`

Then in order to get the `all-clusters-app` to issue the Asan error, from `chip-tool` do:
```
$ ./out/debug/standalone/chip-tool pairing onnetwork 0x12344321 20202021
$ ./out/debug/standalone/chip-tool administratorcommissioning open-basic-commissioning-window 120 0x12344321 0 --timedInteractionTimeoutMs 1000
$  ./out/debug/standalone/chip-tool pairing onnetwork 0x12344321 20202021 --commissioner-name beta
$  ./out/debug/standalone/chip-tool administratorcommissioning open-basic-commissioning-window 120 0x12344321 0 --timedInteractionTimeoutMs 1000
$ ./out/debug/standalone/chip-tool pairing onnetwork 0x12344321 20202021 --commissioner-name gamma
```

woody-apple, 2022-11-02:
Issue Scrub: Assigning to @vivien-apple to confirm this is still happening.

Alami-Amine, 2025-08-25:
I have tried reproducing on Linux and not able to reproduce it (also note that minimum commissioning timeout is now 180 instead of 120)
```bash
./chip-tool pairing onnetwork 0x12344321 20202021
./chip-tool administratorcommissioning open-basic-commissioning-window 180 0x12344321 0 --timedInteractionTimeoutMs 1000
./chip-tool pairing onnetwork 0x12344321 20202021 --commissioner-name beta
./chip-tool administratorcommissioning open-basic-commissioning-window 180 0x12344321 0 --timedInteractionTimeoutMs 1000
./chip-tool pairing onnetwork 0x12344321 20202021 --commissioner-name gamma
```