project-chip/connectedhomeip #15685: Fix error-handling around AddResponseData

bzbarsky-apple, 2022-03-01:
#### Problem
Right now we have various code that does:
1. Call CommandHandler::AddResponseData
2. If that fails add an error status.

But if AddResponseData failed the TLV is going to be in some broken state, so this error-handling code is just wrong.

 #### Proposed Solution
Move adding on error status into AddResponseData, remove it from all consumers.

erjiaqing, 2022-03-07:
I would suggest making the encoding logic "atomic" (i.e. rollback on failure, just like what we do for attributes) instead of encode a status silently, and then provide a wrapper AddResponseDataOrStatus (just like CheckSuccess in `src/app/clusters/general-commissioning-server/general-commissioning-server.cpp`) for "try to add a response, on failure, encode a status". Since we are expecting (or even assuming) the AddResponseData to success.

After checking the related code, the of error code for AddResponseData could be one of:
1. We are running out of buffer pool, in this case, rollback + encode status will also fail in this case.
2. We are trying to encode something that is too big to fit into the response, there could be some bug when preparing the data since we are assuming the data can fir into the response.

bzbarsky-apple, 2022-03-08:
> and then provide a wrapper AddResponseDataOrStatus 

Which do we expect consumers to use most?  We should have AddResponse have the behavior most consumers want, and have a longer name for the less-commonly-desired behavior....

> We are running out of buffer pool, in this case, rollback + encode status will also fail in this case.

Agreed.

> We are trying to encode something that is too big to fit into the response, there could be some bug when preparing the data since we are assuming the data can fir into the response.

There can absolutely be bugs in the cluster impl, or bugs in the client (sending multiple commands in a single invoke whose response commands don't all fit in one packet).

I'm not actually sure what the best option is for the latter case....  If we get to a point we can't even encode a status in the space left, what should we do?

bzbarsky-apple, 2022-03-08:
@Cam @mrjerryjohns thoughts on the above?

mrjerryjohns, 2022-03-08:
I'd agree with updating `AddResponse` to internally, blit a status if it can on failure. But it's still possible that some steps here could fail, in which case, it stands to reason that `AddResponse` should only return an error if it encountered an error in the fallback path of encoding the status.

mrjerryjohns, 2022-03-08:
More-over, trying to add a status after that should actually fail to prevent double encoding of a status.

erjiaqing, 2022-03-15:
> More-over, trying to add a status after that should actually fail to prevent double encoding of a status.

The internal state of command handler will reject encoding more status, also, when AddStatus failed, the command handler cannot encode anything probably.

PR #15906 (2022-03-29): [IM] Make CommandHandler::AddResponseData atomic
#### Problem

Fixes #15685 

CommandHandler::AddResponseData is not atomic, then we cannot encode status since the backed buffer might be corrupted and we are in a wrong state.

#### Change overview
- Adds `ResetResponse`
- Make AddResponseData atomic
- Add AddResponseDataOrFailureStatus

#### Testing
- Adds `TestCommandHandlerCommandEncodeFailure`, and `TestCommandHandlerCommandEncodeExternalFailure`
closed 2022-03-29 by bzbarsky-apple