project-chip/connectedhomeip #15740: [findUserIndexByCredential] Unable to get user: app error [userIndex=1] seen when we perform lock/ unlock operation on M5 from iOS

vinay-apple, 2022-03-02:
#### Problem

[findUserIndexByCredential] Unable to get user: app error [userIndex=1] seen when we perform lock/ unlock operation on M5 from iOS

Steps: 

1. Pair M5 lock device with iOS Home app. 
2. Perform lock/ unlock operation on iOS Home app. 
3. Seeing below error on M5 side when we perform lock/ unlock operation on Home app.

E (612248) chip[ZCL]: [findUserIndexByCredential] Unable to get user: app error [userIndex=1]

Snippet from M5 logs:

I (612248) chip[ZCL]: Received command: LockDoor
E (612248) chip[ZCL]: [findUserIndexByCredential] Unable to get user: app error [userIndex=1]
I (612268) chip[EVL]: Copy Event to next buffer with priority 1
I (612268) chip[DMG]: <RE> Buffer overfilled CHIP_CONFIG_EVENT_LOGGING_BYTE_THRESHOLD 512, schedule engine run
I (612278) chip[IN]: Prepared secure message 0x3ffca0e4 to 0xF88C832ACCDCFE8A (1)  of type 0x9 and protocolId (0, 1) on exchange 9331r with MessageCounter:11537629.
I (612298) chip[IN]: Sending encrypted msg 0x3ffca0e4 with MessageCounter:11537629 to 0xF88C832ACCDCFE8A (1) at monotonic time: 611447 msec
I (612348) chip[EM]: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:15824301 on exchange 9331r
I (616028) chip[EM]: Received message of type 0xa with protocolId (0, 1) and MessageCounter:15824302 on exchange 9332r
I (616028) chip[IN]: Prepared secure message 0x3ffca0e4 to 0xF88C832ACCDCFE8A (1)  of type 0x1 and protocolId (0, 1) on exchange 9332r with MessageCounter:11537630.
I (616038) chip[IN]: Sending encrypted msg 0x3ffca0e4 with MessageCounter:11537630 to 0xF88C832ACCDCFE8A (1) at monotonic time: 615195 msec
I (616118) chip[EM]: Received message of type 0x8 with protocolId (0, 1) and MessageCounter:15824303 on exchange 9332r
I (616128) chip[DMG]: AccessControl: checking f=1 a=c s=0xF88C832ACCDCFE8A t= c=0x0000_0101 e=1 p=o

I (616138) chip[ZCL]: Received command: UnlockDoor
E (616138) chip[ZCL]: [findUserIndexByCredential] Unable to get user: app error [userIndex=1]

[Lock-M5-logs.txt](https://github.com/project-chip/connectedhomeip/files/8172572/Lock-M5-logs.txt)

bzbarsky-apple, 2022-05-21:
@vinay-apple Is this still being a problem on current tip?

woody-apple, 2022-05-26:
Spec Review: @cjandhyala can you please follow up here to see if this is still happening?

cjandhyala, 2022-05-27:
@vinay-apple Could you reproduce this on latest ToT ?