project-chip/connectedhomeip #15753: InteractionModelEngine.cpp:149: CHIP Error 0x0000000B: No memory', reason: '(null)'

woody-apple, 2022-03-02:
#### Problem
Out of memory crash here:
```
Exception Type:  EXC_CRASH
Exception Codes: 0x0000000000000000, 0x0000000000000000
Exception Note:  EXC_CORPSE_NOTIFY
Triggered by Thread:  4

CoreFoundation: *** Terminating app due to uncaught exception 'PFvPvRKN4chip3app9DataModel13DecodableListItEEE: ../../../../../../../../XXX/XXX/connectedhomeip/src/app/InteractionModelEngine.cpp:149: CHIP Error 0x0000000B: No memory', reason: '(null)'

libc++abi.dylib: terminating with uncaught exception of type NSException

libsystem_c.dylib: abort() called

Last Exception Backtrace:
0   CoreFoundation                	       0x1be2c00a8 __exceptionPreprocess + 220
1   libobjc.A.dylib               	       0x1bdb87320 objc_exception_throw + 60
2   XXX                          	       0x1e41d7f1c CHIPCallbackBridge<void*)(void*, chip::app::DataModel::DecodableList<unsigned short> const&)>::CHIPCallbackBridge(NSObject<OS_dispatch_queue>*, void*, NSError*) block_pointer, chip::ChipError*, chip::Callback::Cancelable*) block_pointer, void*)(void*, chip::app::DataModel::DecodableList<unsigned short> const&), bool) + 752
3   XXX                          	       0x1e411f3f4 -[XXX XXX:] + 192
4   XXX                    	       0x1e9a585c4 -[XXX XXX:] + 396
5   XXX                    	       0x1e9a56064 -[XXX XXX:] + 464
6   XXX                    	       0x1e9a68a40 __72-[XXX XXX:]_block_invoke + 352
7   XXX                          	       0x1e408b1c0 ___ZN26CHIPDeviceConnectionBridge11OnConnectedEPvPN4chip22OperationalDeviceProxyE_block_invoke + 44
8   libdispatch.dylib             	       0x1bdc3f048 _dispatch_call_block_and_release + 32
9   libdispatch.dylib             	       0x1bdc40c08 _dispatch_client_callout + 20
10  libdispatch.dylib             	       0x1bdc4834c _dispatch_lane_serial_drain + 668
11  libdispatch.dylib             	       0x1bdc48edc _dispatch_lane_invoke + 444
12  libdispatch.dylib             	       0x1bdc536ec _dispatch_workloop_worker_thread + 648
13  libsystem_pthread.dylib       	       0x1bde6625c _pthread_wqthread + 288
14  libsystem_pthread.dylib       	       0x1bde6d5f8 start_wqthread + 8

```

 #### Proposed Solution
<suggested fix, suggested enhancement>

mrjerryjohns, 2022-03-03:
The line number for the error does not match up with anything on master right now. Is this from a TE* branch?

Also, the back trace indicates some kind of call flow in `DecodableList`, so not sure how `InteractionModelEngine` plays a part in there...

@bzbarsky-apple for more context...

mrjerryjohns, 2022-03-03:
@woody-apple if there are steps to repro, I can do this on my Mac.

bzbarsky-apple, 2022-03-03:
> Is this from a TE* branch?

@sagar-apple @woody-apple what SHA is this from?

terafin, 2022-03-03:
I believe this is from a TE7.5 SHA. @kean-apple Any ideas here?

woody-apple, 2022-03-03:
I believe this is from a TE7.5 SHA. @kean-apple Any ideas here?

kean-apple, 2022-03-04:
I think its fairly old..likely just after TE7 - 83aac592cedfa04cb2941bb0f75f2a4c44ddf799

bzbarsky-apple, 2022-03-04:
OK, so that error is on https://github.com/project-chip/connectedhomeip/blob/83aac592cedfa04cb2941bb0f75f2a4c44ddf799/src/app/InteractionModelEngine.cpp#L149 which is the "we tried to allocate a read client and the pool was full.

Which is not a problem anymore, because we no longer pool-allocate ReadClients.

And on that same rev, the exception being thrown was coming from https://github.com/project-chip/connectedhomeip/blob/83aac592cedfa04cb2941bb0f75f2a4c44ddf799/src/darwin/Framework/CHIP/CHIPCallbackBridgeBase_internal.h#L49:
```
        if (CHIP_NO_ERROR != err) {
            dispatch_async(queue, ^{
                handler(nil, [CHIPError errorForCHIPErrorCode:err]);
            });

            NSString * errorStr = [NSString stringWithFormat:@"%s: %s", typeid(T).name(), chip::ErrorStr(err)];
            @throw [NSException exceptionWithName:errorStr reason:nil userInfo:nil];
```
which is also gone.  Now we do:
```
        if (CHIP_NO_ERROR != err) {
            NSLog(@"Failure performing action. C++-mangled success callback type: '%s', error: %s", typeid(T).name(),
                chip::ErrorStr(err));

            // Take the normal async error-reporting codepath.  This will also
            // handle cleaning us up properly.
            DispatchFailure(this, [CHIPError errorForCHIPErrorCode:err]);
        }
```
and report the error async, without doing that @throw business.

closed 2022-03-04 by bzbarsky-apple