project-chip/connectedhomeip #15761: Lost access to accessory after controller reboot

sagar-apple, 2022-03-03:
#### Problem
1. Run all-clusters-app in a shell
2. Commission the app using the chip-tool cli
3. Open the commissioning window (sample command below)
```
chip-tool pairing  open-commissioning-window <node-id> 1 600 1000 2280
```
4. Copy the manual code from the command above and paste it into the QRCode scanner screen on iOS CHIPTool.
5. Verify iOS CHIPTool was able to commission the accessory (try toggle or enumeration...)
6. Restart iOS CHIPTool. 
7. Verify that control still works (toggle/enumeration) - FAIL

After restarting the iOS application, the accessory denies access control to the iOS CHIPTool controller.  
 
 ```
 [1646267292692] [59542:42704697] CHIP: [EM] Handling via exchange: 23826r, Delegate: 0x1033633c0
[1646267292692] [59542:42704697] CHIP: [DMG] InvokeRequestMessage =
[1646267292692] [59542:42704697] CHIP: [DMG] {
[1646267292692] [59542:42704697] CHIP: [DMG] 	suppressResponse = false, 
[1646267292692] [59542:42704697] CHIP: [DMG] 	timedRequest = false, 
[1646267292692] [59542:42704697] CHIP: [DMG] 	InvokeRequests =
[1646267292692] [59542:42704697] CHIP: [DMG] 	[
[1646267292692] [59542:42704697] CHIP: [DMG] 		CommandDataIB =
[1646267292692] [59542:42704697] CHIP: [DMG] 		{
[1646267292692] [59542:42704697] CHIP: [DMG] 			CommandPathIB =
[1646267292692] [59542:42704697] CHIP: [DMG] 			{
[1646267292692] [59542:42704697] CHIP: [DMG] 				EndpointId = 0x1,
[1646267292692] [59542:42704697] CHIP: [DMG] 				ClusterId = 0x6,
[1646267292692] [59542:42704697] CHIP: [DMG] 				CommandId = 0x2,
[1646267292692] [59542:42704697] CHIP: [DMG] 			},
[1646267292692] [59542:42704697] CHIP: [DMG] 			
[1646267292692] [59542:42704697] CHIP: [DMG] 			CommandData = 
[1646267292692] [59542:42704697] CHIP: [DMG] 			{
[1646267292692] [59542:42704697] CHIP: [DMG] 			},
[1646267292692] [59542:42704697] CHIP: [DMG] 		},
[1646267292692] [59542:42704697] CHIP: [DMG] 		
[1646267292692] [59542:42704697] CHIP: [DMG] 	],
[1646267292692] [59542:42704697] CHIP: [DMG] 	
[1646267292692] [59542:42704697] CHIP: [DMG] 	InteractionModelRevision = 1
[1646267292692] [59542:42704697] CHIP: [DMG] },
[1646267292692] [59542:42704697] CHIP: [DMG] AccessControl: checking f=3 a=c s=0xC9920843139F706A t= c=0x0000_0006 e=1 p=o
[1646267292692] [59542:42704697] CHIP: [DMG] AccessControl: denied
```

kean-apple, 2022-03-03:
This is seen with a single admin case as well:
1. Pair M5 board via iOS chiptool
2. Verify able to control M5 board eg. toggle light
3. Force close iOS chiptool 
4. Relaunch iOS chiptool - toggling light fails with AccessControl: denied

bzbarsky-apple, 2022-03-03:
This is a regression from https://github.com/project-chip/connectedhomeip/pull/15729

When we normally bring up the Matter stack on iOS chip-tool, we do this:
```
src/darwin/CHIPTool/CHIPTool/Framework Helpers/DefaultsUtils.m:        [controller startup:storage vendorId:0 nocSigner:nil];
```
where `storage` is a `CHIPToolPersistentStorageDelegate` instance.

The restart added in #15729, though, does:
```
src/darwin/CHIPTool/CHIPTool/View Controllers/QRCode/QRCodeViewController.m:    [self.chipController startup:nil vendorId:0 nocSigner:nil];
```
and then we end up using a `NSUserDefaults` as the storage.  The different storages can and do end up with different node ids.

So now the order of events is:
1. Start iOS chiptool. 
2. Go into the "QRCode scanner" screen.  Brings up Matter stack, we're using the `CHIPToolPersistentStorageDelegate` as storage, have node id X.
3. Tap the camera icon.  This does the stack restart, comes up with the other storage, node id becomes Y.
4. We commission while we are node id Y and install the ACLs for that.
5. Restart the iOS chiptool.  Try to do something that brings up the Matter stack.  It comes up with node id X again, and we fail ACL checks.

closed 2022-03-03 by andy31415