project-chip/connectedhomeip #15766: After M5Stack crashes, subscriptions are not working

woody-apple, 2022-03-03:
#### Problem
After the M5 stack crashes, all subscriptions fail

[device-unreachable-after-M5-crashes.txt](https://github.com/project-chip/connectedhomeip/files/8175015/device-unreachable-after-M5-crashes.txt)

mrjerryjohns, 2022-03-03:
This is to be expected, given that the client side continues to behave as though the server (the M5Stack) still has a valid CASE session, which it doesn't.

The only way to recover from this is for the subscription to eventually time out, triggering a potential re-establishment of CASE.

mrjerryjohns, 2022-03-03:
Alternatively, I'd think the controller/client would see the advertisement of the operational node upon boot, triggering re-establishment of CASE?

mrjerryjohns, 2022-03-03:
@woody-apple do you have client side logs?

terafin, 2022-03-03:
@kean-apple ?

terafin, 2022-03-03:
@mrjerryjohns note: shouldn't this have auto re-established successfully? (assuming devices were awake)

woody-apple, 2022-03-03:
@kean-apple?

woody-apple, 2022-03-03:
Note: @mrjerryjohns Shouldn't this have [if devices were awake] reconnected?

kean-apple, 2022-03-03:
will have to gather client side logs. I presume this is same issue as reported in https://github.com/project-chip/connectedhomeip/issues/15283

kean-apple, 2022-03-03:
Seems like general communication is broken once the M5 board reboots

SHA:  3b834619186a84db03281a5c76f236562db27ebd
1. Pair M5 board with iOS chiptool app - verify toggle light works
2. Press red button on M5 board to reboot board 
3. Try to control light on M5 board - get error 0x01

Attached M5 and chiptool client logs

[unable-to-control-M5-board-after-reboot-chiptool-logs.txt](https://github.com/project-chip/connectedhomeip/files/8180324/unable-to-control-M5-board-after-reboot-chiptool-logs.txt)
[unable-to-control-M5-board-after-reboot-M5logs.txt](https://github.com/project-chip/connectedhomeip/files/8180325/unable-to-control-M5-board-after-reboot-M5logs.txt)

bzbarsky-apple, 2022-03-03:
Is this because iOS CHIPTool keeps trying to use the now-stale CASE session?  @kean-apple @sagar-apple

sagar-apple, 2022-03-03:
I think it's because there's no retry for subscriptions in CHIPTool.

kean-apple, 2022-03-15:
@erjiaqing any update on this issue? this issue blocks any stress testing & living-on/dog fooding for Matter devices

kean-apple, 2022-03-17:
yes I agree. I'll transfer the smoke test blocker label to that issue

yunhanw-google, 2022-08-10:
@kean-apple  is this still issue?

kean-apple, 2022-08-10:
this is fixed now

closed 2022-08-10 by kean-apple