project-chip/connectedhomeip #15767: M5 lightbulb pairing fails when 11-digit Manual setup code is used to pair

woody-apple, 2022-03-03:
#### Problem

Pairing error occurred

Logs:
```
rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6628
load:0x40078000,len:14780
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v4.4-263-g000d3823bb-dirty 2nd stage bootloader
I (27) boot: compile time 12:01:19
I (27) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (84) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (91) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (99) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (106) boot:  5 ot_storage       Unknown data     01 3a 00317000 00002000
I (114) boot: End of partition table
I (118) boot_comm: chip revision: 3, min. application chip revision: 0
I (125) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=36fd0h (225232) map
I (215) esp_image: segment 1: paddr=00056ff8 vaddr=3ffbdb60 size=08518h ( 34072) load
I (229) esp_image: segment 2: paddr=0005f518 vaddr=40080000 size=00b00h (  2816) load
I (230) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=f48c8h (1001672) map
I (597) esp_image: segment 4: paddr=001548f0 vaddr=40080b00 size=1d8d8h (121048) load
I (647) esp_image: segment 5: paddr=001721d0 vaddr=50000000 size=00010h (    16) load
I (663) boot: Loaded app from partition at offset 0x20000
I (663) boot: Disabling RNG early entropy source...
I (675) cpu_start: Pro cpu up.
I (675) cpu_start: Starting app cpu, entry point is 0x400814dc
0x400814dc: call_start_cpu1 at /Users/ajaybadagowni/tools/esp-idf/components/esp_system/port/cpu_start.c:160

I (662) cpu_start: App cpu up.
I (691) cpu_start: Pro cpu start user code
I (691) cpu_start: cpu freq: 160000000
I (691) cpu_start: Application information:
I (696) cpu_start: Project name:     chip-all-clusters-app
I (702) cpu_start: App version:      TE8/rc3-140-g448578dd0-dirty
I (709) cpu_start: Compile time:     Mar  1 2022 18:02:47
I (715) cpu_start: ELF file SHA256:  89c08025d8cbfbeb...
I (721) cpu_start: ESP-IDF:          v4.4-263-g000d3823bb-dirty
I (728) heap_init: Initializing. RAM available for dynamic allocation:
I (734) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (741) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (747) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (753) heap_init: At 3FFD6A80 len 00009580 (37 KiB): DRAM
I (759) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (765) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (772) heap_init: At 4009E3D8 len 00001C28 (7 KiB): IRAM
I (779) spi_flash: detected chip: generic
I (783) spi_flash: flash io: dio
W (786) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (803) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (810) all-clusters-app: All Clusters Demo!
I (820) all-clusters-app: This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, 
[... 654 lines cut ...]
I (106280) chip[ZCL]: OpCreds: Call to fabricListChanged
I (106280) chip[DIS]: CHIP minimal mDNS started advertising.
I (106300) chip[DIS]: Failed to find a valid admin pairing. Node ID unknown
I (106300) chip[DIS]: Start dns-sd server - no current nodeId
I (106320) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32769 discriminator=3674/14
I (106320) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'.
I (106340) chip[DIS]: mDNS service published: _matterc._udp
I (106340) chip[ZCL]: OpCreds: Call to fabricListChanged
I (106350) chip[IN]: Prepared secure message 0x3ffe7754 to 0xC28EBF23EE21840D (1)  of type 0x9 and protocolId (0, 1) on exchange 45569r with MessageCounter:16456346.
I (106360) chip[IN]: Sending encrypted msg 0x3ffe7754 with MessageCounter:16456346 to 0xC28EBF23EE21840D (1) at monotonic time: 105582 msec
E (106380) chip[DL]: Long dispatch time: 227 ms, for event type 3
E (106390) chip[IN]: Data received on an unknown session (LSID=2). Dropping it!
E (106390) chip[IN]: Data received on an unknown session (LSID=2). Dropping it!
I (106510) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 0 ) indicate = 0 notify = 0
E (106520) chip[BLE]: no endpoint for unsub recvd
I (106520) chip[DL]: CHIPoBLE unsubscribe received
I (106530) chip[DL]: BLE GAP connection terminated (con 0 reason 0x213)
I (106530) app-devicecallbacks: CHIPoBLE disconnected
I (106550) app-devicecallbacks: Current free heap: 67112

E (113390) chip[IN]: Data received on an unknown session (LSID=3). Dropping it!
E (118410) chip[IN]: Data received on an unknown session (LSID=3). Dropping it!
E (118720) chip[IN]: Data received on an unknown session (LSID=3). Dropping it!
E (119030) chip[IN]: Data received on an unknown session (LSID=3). Dropping it!
```

shubhamdp, 2022-07-14:
@dhrishi tried with `examples/temperature-measurement-app/esp32` with manual code and commissioning completed with success.
[temp-app-esp32.log](https://github.com/project-chip/connectedhomeip/files/9110631/temp-app-esp32.log)
[chip-tool.log](https://github.com/project-chip/connectedhomeip/files/9110633/chip-tool.log)

woody-apple, 2022-11-02:
Issue Scrub: Closing this given it sounds like this is confirmed working.

closed 2022-11-02 by woody-apple