project-chip/connectedhomeip #15776: Door Lock Pairing Issue

sethunk, 2022-03-03:
**Problem**
  Pairing of door lock is not working.

**Logs:**
[DoorLock.txt](https://github.com/project-chip/connectedhomeip/files/8175905/DoorLock.txt)

cjandhyala, 2022-03-10:
@sethunk The attached log shows pairing is fine.. Can you attach the correct log ?

sethunk, 2022-03-11:
Have recaptured the logs and attached the logs with today date

Pls note that in the previous logs also commands are not working.

sethunk, 2022-03-11:
Have recaptured the logs and attached the logs with today date

Pls note that in the previous logs also commands are not working.

In door lock pairing, 
Scenario -1 : With ble, pairing is not working.
Scenario -2 : With ethernet, it is commissioned with a particular endpoint (eg: 0)  but when we execute the attributes for that endpoint, we get the response as 0xc3-Unsupported Cluster.
		       When we execute the attributes with other endpoint (eg: 1) , we get the response for that attribute.

Logs:
ble
[Doorlock_cntrl_11.03.2022.txt](https://github.com/project-chip/connectedhomeip/files/8230736/Doorlock_cntrl_11.03.2022.txt)
[Doorlock_dut_11.03.2022.txt](https://github.com/project-chip/connectedhomeip/files/8230738/Doorlock_dut_11.03.2022.txt)

ethernet
[Doorlock_cntrl-ethernet_11.03.2022.txt](https://github.com/project-chip/connectedhomeip/files/8230929/Doorlock_cntrl-ethernet_11.03.2022.txt)
[Doorlock_dut-ethernet_11.03.2022.txt](https://github.com/project-chip/connectedhomeip/files/8230930/Doorlock_dut-ethernet_11.03.2022.txt)

Survensa, 2022-04-05:
Still not able to pair DUT with Cntrl for door-lock app with out "`main.cpp`" work around.

Commit used : 41a431dcc7c123699818636afd01b57412a6f3a4		

Error Issue :

```
ubuntu@matter-7:~/cntrl/connectedhomeip/examples/chip-tool/out/debug$ ./chip-tool pairing ble-wifi 1 chipsetup3 matter123 20202021 3840
[1649147087.797286][17465:17465] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1649147087.797609][17465:17465] CHIP:DL: writing settings to file (/tmp/chip_kvs-ddt2XS)
[1649147087.797816][17465:17465] CHIP:DL: renamed tmp file to file (/tmp/chip_kvs)
[1649147087.798227][17465:17465] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1649147087.798380][17465:17465] CHIP:DL: writing settings to file (/tmp/chip_factory.ini-yxkW9A)
[1649147087.798524][17465:17465] CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
[1649147087.798610][17465:17465] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1649147087.798776][17465:17465] CHIP:DL: writing settings to file (/tmp/chip_config.ini-srL8vC)
[1649147087.798912][17465:17465] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1649147087.798995][17465:17465] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1649147087.799127][17465:17465] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-iT43hS)
[1649147087.799251][17465:17465] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1649147087.799511][17465:17465] CHIP:DL: writing settings to file (/tmp/chip_factory.ini-jl1qBS)
[1649147087.800049][17465:17465] CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
[1649147087.800103][17465:17465] CHIP:DL: NVS set: chip-factory/unique-id = "33326BDBE2C7802A"
[1649147087.800260][17465:17465] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-INYUVb)
[1649147087.800644][17465:17465] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1649147087.800690][17465:17465] CHIP:DL: NVS set: chip-counters/reboot-count = 1 (0x1)
[1649147087.800832][17465:17465] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-Tob45T)
[1649147087.802224][17465:17465] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1649147087.802277][17465:17465] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1649147087.802436][17465:17465] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-gh4LiN)
[1649147087.803476][17465:17465] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1649147087.803535][17465:17465] CHIP:DL: NVS set: chip-counters/boot-reason = 0 (0x0)
[1649147087.803700][17465:17465] CHIP:DL: writing settings to file (/tmp/chip_config.ini-HWuFb2)
[1649147087.804085][17465:17465] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1649147087.804127][17465:17465] CHIP:DL: NVS set: chip-config/regulatory-location = 0 (0x0)
[1649147087.804258][17465:17465] CHIP:DL: writing settings to file (/tmp/chip_config.ini-es9Tdd)
[1649147087.806086][17465:17465] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1649147087.806175][17465:17465] CHIP:DL: NVS set: chip-config/location-capability = 0 (0x0)
[1649147087.807470][17465:17465] CHIP:DL: Got Ethernet interface: eth0
[1649147087.808250][17465:17465] CHIP:DL: Found the primary Ethernet interface:eth0
[1649147087.809028][17465:17465] CHIP:DL: Got WiFi interface: wlan0
[1649147087.809121][17465:17465] CHIP:DL: Failed to reset WiFi statistic counts
[1649147087.809219][17465:17465] CHIP:IN: UDP::Init bind&listen port=0
[1649147087.809383][17465:17465] CHIP:IN: UDP::Init bound to port=35354
[1649147087.809417][17465:17465] CHIP:IN: BLEBase::Init - setting/overriding transport
[1649147087.809445][17465:17465] CHIP:IN: TransportMgr initialized
[1649147087.809489][17465:17465] CHIP:DIS: Init fabric pairing table with server storage
[1649147087.809546][17465:17465] CHIP:DIS: Add fabric pairing table delegate
[1649147087.809856][17465:17465] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-Zrh6vW)
[1649147087.810609][17465:17465] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1649147087.810721][17465:17465] CHIP:DL: NVS set: chip-counters/GlobalMCTR = 1000 (0x3E8)
[1649147087.813278][17465:17465] CHIP:ZCL: Using ZAP configuration...
[1649147087.818479][17465:17465] CHIP:DL: Avahi client registered
[1649147087.820111][17465:17465] CHIP:CTL: System State Initialized...
[1649147087.820239][17465:17465] CHIP:CTL: Setting attestation nonce to random value
[1649147087.820309][17465:17465] CHIP:CTL: Setting CSR nonce to random value
[1649147087.820407][17465:17465] CHIP:IN: UDP::Init bind&listen port=5550
[1649147087.820582][17465:17465] CHIP:IN: UDP::Init bound to port=5550
[1649147087.820616][17465:17465] CHIP:IN: TransportMgr initialized
[1649147087.820719][17465:17465] CHIP:CTL: Setting attestation nonce to random value
[1649147087.820788][17465:17465] CHIP:CTL: Setting CSR nonce to random value
[1649147087.821334][17465:17465] CHIP:CTL: Couldn't get ExampleOpCredsCAKey from storage: ../../config/PersistentStorage.cpp:108: CHIP Error 0x000000A0: Value not found in the persisted storage
[1649147087.825207][17465:17465] CHIP:CTL: Couldn't get ExampleOpCredsICAKey from storage: ../../config/PersistentStorage.cpp:108: CHIP Error 0x000000A0: Value not found in the persisted storage
[1649147087.827203][17465:17465] CHIP:CTL: Generating RCAC
[1649147087.828583][17465:17465] CHIP:CTL: Generating ICAC
[... cut ...]
[1649147099.512861][17465:17465] CHIP:BLE: Auto-closing end point's BLE connection.
[1649147099.512914][17465:17465] CHIP:DL: Closing BLE GATT connection (con 0xffff6c0600f0)
[1649147099.513025][17465:17468] CHIP:DL: BluezDisconnect peer=E4:5F:01:30:E6:82
[1649147099.514767][17465:17465] CHIP:DL: System Layer shutdown
[1649147099.514974][17465:17465] CHIP:TOO: Run command failure: ../../third_party/connectedhomeip/src/controller/AutoCommissioner.cpp:167: CHIP Error 0x0000002F: Invalid argument
```

vivien-apple, 2022-04-05:
Well if you read the log attentively you can see that those lines:
```
[1649147099.505716][17465:17471] CHIP:CTL: Required network information not provided in commissioning parameters
[1649147099.505766][17465:17471] CHIP:CTL: Parameters supplied: wifi (yes) thread (no)
[1649147099.505815][17465:17471] CHIP:CTL: Device supports: wifi (no) thread(yes)
```

Basically it states that the device does not support `WiFi`, so it makes sense that passing over `wifi` credentials does not work.

vivien-apple, 2022-04-05:
"Scenario -2 : With ethernet, it is commissioned with a particular endpoint (eg: 0) but when we execute the attributes for that endpoint, we get the response as 0xc3-Unsupported Cluster."

Is it expected that this device has a "Door Lock" cluster onto endpoint "0" ? If it works on "1" it just seems correct to me.

Survensa, 2022-04-05:
@vivien-apple Door lock works fine with `wifi` but only after a workaround in `main.cpp` file.

Expectation : Chiptool must have this fix default for doorlock

Pairing log after " main.cpp" workaround :

```
ubuntu@ubuntu:~/sethu-apr1/connectedhomeip/examples/chip-tool/out/debug$ ./chip-tool pairing ble-wifi 1 GRLPrivate_EXT matter123 20202021 3840
[1648813479.801267][11968:11968] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1648813479.801553][11968:11968] CHIP:DL: writing settings to file (/tmp/chip_kvs-Tm4Gsh)
[1648813479.801764][11968:11968] CHIP:DL: renamed tmp file to file (/tmp/chip_kvs)
[1648813479.802195][11968:11968] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1648813479.802341][11968:11968] CHIP:DL: writing settings to file (/tmp/chip_factory.ini-QN89af)
[1648813479.802481][11968:11968] CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
[1648813479.802569][11968:11968] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1648813479.802689][11968:11968] CHIP:DL: writing settings to file (/tmp/chip_config.ini-MReYHZ)
[1648813479.802812][11968:11968] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1648813479.802891][11968:11968] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1648813479.802996][11968:11968] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-LAlLLe)
[1648813479.803114][11968:11968] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1648813479.803326][11968:11968] CHIP:DL: writing settings to file (/tmp/chip_factory.ini-9taG9G)
[1648813479.803837][11968:11968] CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
[1648813479.803884][11968:11968] CHIP:DL: NVS set: chip-factory/unique-id = "233E5FEEDDB5A43A"
[1648813479.804020][11968:11968] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-9mBofW)
[1648813479.804357][11968:11968] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1648813479.804400][11968:11968] CHIP:DL: NVS set: chip-counters/reboot-count = 1 (0x1)
[1648813479.804540][11968:11968] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-dUq19I)
[1648813479.805499][11968:11968] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1648813479.805560][11968:11968] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1648813479.805781][11968:11968] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-twOPZk)
[1648813479.806395][11968:11968] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1648813479.806449][11968:11968] CHIP:DL: NVS set: chip-counters/boot-reason = 0 (0x0)
[1648813479.806594][11968:11968] CHIP:DL: writing settings to file (/tmp/chip_config.ini-4suQSi)
[1648813479.807026][11968:11968] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1648813479.807075][11968:11968] CHIP:DL: NVS set: chip-config/regulatory-location = 0 (0x0)
[1648813479.807208][11968:11968] CHIP:DL: writing settings to file (/tmp/chip_config.ini-U3vaWE)
[1648813479.807763][11968:11968] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1648813479.807810][11968:11968] CHIP:DL: NVS set: chip-config/location-capability = 0 (0x0)
[1648813479.808788][11968:11968] CHIP:DL: Got Ethernet interface: eth0
[1648813479.809394][11968:11968] CHIP:DL: Found the primary Ethernet interface:eth0
[1648813479.810048][11968:11968] CHIP:DL: Got WiFi interface: wlan0
[1648813479.810122][11968:11968] CHIP:DL: Failed to reset WiFi statistic counts
[1648813479.810266][11968:11968] CHIP:IN: UDP::Init bind&listen port=0
[1648813479.810423][11968:11968] CHIP:IN: UDP::Init bound to port=43833
[1648813479.810450][11968:11968] CHIP:IN: BLEBase::Init - setting/overriding transport
[1648813479.810472][11968:11968] CHIP:IN: TransportMgr initialized
[1648813479.810508][11968:11968] CHIP:DIS: Init fabric pairing table with server storage
[1648813479.810556][11968:11968] CHIP:DIS: Add fabric pairing table delegate
[1648813479.810765][11968:11968] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-JHS1Zt)
[1648813479.811360][11968:11968] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1648813479.811415][11968:11968] CHIP:DL: NVS set: chip-counters/GlobalMCTR = 1000 (0x3E8)
[1648813479.812944][11968:11968] CHIP:ZCL: Using ZAP configuration...
[1648813479.817541][11968:11968] CHIP:DL: Avahi client registered
[1648813479.818878][11968:11968] CHIP:CTL: System State Initialized...
[1648813479.818974][11968:11968] CHIP:CTL: Setting attestation nonce to random value
[1648813479.819028][11968:11968] CHIP:CTL: Setting CSR nonce to random value
[1648813479.819108][11968:11968] CHIP:IN: UDP::Init bind&listen port=5550
[1648813479.819254][11968:11968] CHIP:IN: UDP::Init bound to port=5550
[1648813479.819281][11968:11968] CHIP:IN: TransportMgr initialized
[1648813479.819429][11968:11968] CHIP:CTL: Setting attestation nonce to random value
[1648813479.819482][11968:11968] CHIP:CTL: Setting CSR nonce to random value
[1648813479.819974][11968:11968] CHIP:CTL: Couldn't get ExampleOpCredsCAKey from storage: ../../config/PersistentStorage.cpp:108: CHIP Error 0x000000A0: Value not found in the persisted storage
[1648813479.823052][11968:11968] CHIP:CTL: Couldn't get ExampleOpCredsICAKey from storage: ../../config/PersistentStorage.cpp:108: CHIP Error 0x000000A0: Value not found in the persisted storage
[1648813479.824310][11968:11968] CHIP:CTL: Generating RCAC
[... cut ...]
[1648813492.685010][11968:11968] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1648813492.685062][11968:11968] CHIP:DL: Inet Layer shutdown
[1648813492.685097][11968:11968] CHIP:DL: BLE shutdown
[1648813492.685129][11968:11968] CHIP:BLE: Auto-closing end point's BLE connection.
[1648813492.685159][11968:11968] CHIP:DL: Closing BLE GATT connection (con 0xffffa403f110)
[1648813492.685413][11968:11971] CHIP:DL: BluezDisconnect peer=E4:5F:01:0E:BE:FE
[1648813492.687557][11968:11968] CHIP:DL: System Layer shutdown
```

sethunk, 2022-04-13:
Checked with SQA @Survensa , the team is able to pair doorlock app without any workaround

closed 2022-04-13 by sethunk