project-chip/connectedhomeip #15780: EndPoint Mapping issue - Network commissioning

Sivaramgrl, 2022-03-03:
**### Problem:**
During the commissioning of TH & DUT via WiFi, The Endpoint is set as 0 by default. 
- When executing the command with Endpoint-0. The user could not find the expected output
- When executing the command with Endpoint-1. The user can find the output

**### Expected behavior:**
When DUT is commissioned with TH via WiFi with Endpoint-0, the commands must have the expected output with Endpoint-0
 **### Actual behavior:**
When DUT is commissioned with TH via WiFi with Endpoint-0, the commands must have the expected output with Endpoint-0, but the user can get the expected output on EndPoint-1

**### Command:** 
**- sudo ./chip-tool  networkcommissioning scan-networks null 0 1 0**
[1646293689.990014][2928:2933] CHIP:CTL: Device connected callback with null pairing delegate. Ignoring
[1646293689.990067][2928:2933] CHIP:EM: Sending Standalone Ack for MessageCounter:1871595063 on exchange 3459i
[1646293689.990110][2928:2933] CHIP:IN: Prepared unauthenticated message 0xffffa1bcb938 to 0x0000000000000000 (0)  of type 0x10 and protocolId (0, 0) on exchange 3459i with MessageCounter:525156546.
[1646293689.990157][2928:2933] CHIP:IN: Sending unauthenticated msg 0xffffa1bcb938 with MessageCounter:525156546 to 0x0000000000000000 at monotonic time: 5262853 msec
[1646293689.990248][2928:2933] CHIP:EM: Flushed pending ack for MessageCounter:1871595063 on exchange 3459i
[1646293690.190932][2928:2933] CHIP:EM: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:13758687 on exchange 3460i
[1646293690.191036][2928:2933] CHIP:EM: Found matching exchange: 3460i, Delegate: 0xffff8c003aa0
[1646293690.191120][2928:2933] CHIP:EM: Rxd Ack; Removing MessageCounter:11356051 from Retrans Table on exchange 3460i
[1646293690.191180][2928:2933] CHIP:EM: Removed CHIP MessageCounter:11356051 from RetransTable on exchange 3460i
[1646293699.967781][2928:2928] CHIP:-: ../../examples/chip-tool/commands/common/CHIPCommand.cpp:249: CHIP Error 0x00000032: Timeout at ../../examples/chip-tool/commands/common/CHIPCommand.cpp:62
[1646293699.967832][2928:2928] CHIP:TOO: Run command failure: ../../examples/chip-tool/commands/common/CHIPCommand.cpp:249: CHIP Error 0x00000032: Timeout
[1646293699.976755][2928:2928] CHIP:SPT: VerifyOrDie failure at ../../src/lib/support/Pool.h:304: Allocated() == 0
Aborted

**- sudo ./chip-tool  networkcommissioning scan-networks null 0 1 1**
1646293616.644435][2917:2922] CHIP:DMG: Received Command Response Data, Endpoint=1 Cluster=0x0000_0031 Command=0x0000_0001
[1646293616.644542][2917:2922] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0031 Command 0x0000_0001
[1646293616.644695][2917:2922] CHIP:TOO:   ScanNetworksResponse: {
[1646293616.644775][2917:2922] CHIP:TOO:     networkingStatus: 0
[1646293616.644833][2917:2922] CHIP:TOO:     debugText:
[1646293616.644905][2917:2922] CHIP:TOO:     wiFiScanResults: 1 entries
[1646293616.645064][2917:2922] CHIP:TOO:       [1]: {
[1646293616.645123][2917:2922] CHIP:TOO:         Security: 8
[1646293616.645180][2917:2922] CHIP:TOO:         Ssid: 47524C507269766174655F455854
[1646293616.645236][2917:2922] CHIP:TOO:         Bssid: E01CFCE4B236
[1646293616.645290][2917:2922] CHIP:TOO:         Channel: 11
[1646293616.645343][2917:2922] CHIP:TOO:         WiFiBand: 0
[1646293616.645396][2917:2922] CHIP:TOO:         Rssi: -7
[1646293616.645451][2917:2922] CHIP:TOO:        }
[1646293616.645510][2917:2922] CHIP:TOO:    }
[1646293616.645591][2917:2922] CHIP:DMG: ICR moving to [AwaitingDe]

**### full log:** 
[Endpoint mapping.txt](https://github.com/project-chip/connectedhomeip/files/8175938/Endpoint.mapping.txt)

erjiaqing, 2022-04-14:
What is the platform & application and what is the commit id of server?

After checking all generated .matter IDL files, no applications has network commissioning cluster on endpoint 1

There was a version that WiFi endpoint is configued to exist on endpoint 1, but on ToT, the main endpoint id of network commissioning cluster is endpoint 0, please ensure your checkout is newer than https://github.com/project-chip/connectedhomeip/pull/15067, thanks.

Sivaramgrl, 2022-04-19:
@erjiaqing,
### commit id:
commit 87938704ed93dde93e960605ae6a714923efa2e0
Date:   Mon Apr 18 18:59:53 2022 -0700

### Issue:
We could not execute the scan-networks command in EP0 & EP1. 
As scan-networks command is not having any argument as per spec. ( SSID & Breadcrumb)
So we could not able to validate the issue. 

### Inputs:
the Rpi's are connected via WiFi. Commissioning is on EP0.
Once the scan-networks command is fixed. We can verify this issue. 

### Observations:
1. When the scan-network command is executed with EP0 - DUT responds as ''General error: 0x85 (INVALID_COMMAND)"
2. When the scan-network command is executed with EP1 - DUT responds as "General error: 0xc3 (UNSUPPORTED_CLUSTER)"
3. When the scan-network command is executed it's requests for only 2 arguments 
     i. node-id/group-id
     ii. endpoint-id-ignored-for-group-commands
   we could not able to find the SSID and Breadcrumb arguments

Due to the above one, we could not able to verify the scan-network command with EP0 & EP1.
### Logs:
[Scannetwork_TH-NC.txt](https://github.com/project-chip/connectedhomeip/files/8511755/Scannetwork_TH-NC.txt)
[Scannetwork_DUT -NC.txt](https://github.com/project-chip/connectedhomeip/files/8511756/Scannetwork_DUT.-NC.txt)

erjiaqing, 2022-04-19:
INVALID_COMMAND is throwed before any cluster logic, so I guess the DUT is invoked with wrong command line arguments.

Could you try to read the FeatureMap bits from the network commissioning cluster? I cannot found logs realted to wpa_supplicant from the DUT, when wifi is enabled, wpa_supplicant should be initialized during starting up,

Please ensure the all-cluster-apps is executed with `--wifi` in the command line arguments.

Sivaramgrl, 2022-04-20:
Thanks, @erjiaqing.
As per your comments in slack. 
Now we can execute the command in EP0.

> ubuntu@matter-7:~/cntrl/connectedhomeip/examples/chip-tool/out/debug$ ./chip-tool networkcommissioning scan-networks 1 0 --Ssid null --Breadcrumb 0

The scan-network command with EP1 DUT responds with a General error: 0xc3 (UNSUPPORTED_CLUSTER)

> [1650455853.032393][8346:8352] CHIP:DMG: Received Command Response Status for Endpoint=1 Cluster=0x0000_0031 Command=0x0000_0000 Status=0xc3
> [1650455853.032437][8346:8352] CHIP:TOO: Error: IM Error 0x000005C3: General error: 0xc3 (UNSUPPORTED_CLUSTER)
> [1650455853.032483][8346:8352] CHIP:DMG: ICR moving to [AwaitingDe]
> [1650455853.032553][8346:8352] CHIP:EM: Sending Standalone Ack for MessageCounter:3663209 on exchange 46878i

The scan-network command with EP0 DUT responds with the ScanNetworkResponse command

> [1650455846.948457][8339:8344] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0001
> [1650455846.948515][8339:8344] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Command 0x0000_0001
> [1650455846.948633][8339:8344] CHIP:TOO:   ScanNetworksResponse: {
> [1650455846.948687][8339:8344] CHIP:TOO:     networkingStatus: 0
> [1650455846.948773][8339:8344] CHIP:TOO:     wiFiScanResults: 15 entries
> [1650455846.948852][8339:8344] CHIP:TOO:       [1]: {
> [1650455846.948876][8339:8344] CHIP:TOO:         Security: 12
> [1650455846.948903][8339:8344] CHIP:TOO:         Ssid: 63686970736574757033
> [1650455846.948927][8339:8344] CHIP:TOO:         Bssid: 0C0E764D4144
> [1650455846.948949][8339:8344] CHIP:TOO:         Channel: 11
> [1650455846.948971][8339:8344] CHIP:TOO:         WiFiBand: 0
> [1650455846.948994][8339:8344] CHIP:TOO:         Rssi: -10
> [1650455846.949019][8339:8344] CHIP:TOO:        }
> [1650455846.949053][8339:8344] CHIP:TOO:       [2]: {
> [1650455846.949078][8339:8344] CHIP:TOO:         Security: 8
> [1650455846.949100][8339:8344] CHIP:TOO:         Ssid: 47524C50726976617465
> [1650455846.949123][8339:8344] CHIP:TOO:         Bssid: 6032B197B89F
> [1650455846.949145][8339:8344] CHIP:TOO:         Channel: 153
> [1650455846.949167][8339:8344] CHIP:TOO:         WiFiBand: 2
> [1650455846.949189][8339:8344] CHIP:TOO:         Rssi: -35
> [1650455846.949211][8339:8344] CHIP:TOO:        }
> [1650455846.949243][8339:8344] CHIP:TOO:       [3]: {
> [1650455846.949266][8339:8344] CHIP:TOO:         Security: 8
> [1650455846.949288][8339:8344] CHIP:TOO:         Ssid: 7A6967626565315F3547
> [1650455846.949311][8339:8344] CHIP:TOO:         Bssid: E848B8C2123A
> [1650455846.949332][8339:8344] CHIP:TOO:         Channel: 153
> [1650455846.949354][8339:8344] CHIP:TOO:         WiFiBand: 2
> [1650455846.949375][8339:8344] CHIP:TOO:         Rssi: -43
> [1650455846.949397][8339:8344] CHIP:TOO:        }
> [1650455846.949428][8339:8344] CHIP:TOO:       [4]: {
> [1650455846.949451][8339:8344] CHIP:TOO:         Security: 12
> [1650455846.949474][8339:8344] CHIP:TOO:         Ssid: 7A69676265652D7468726561642D3547487A
> [1650455846.949496][8339:8344] CHIP:TOO:         Bssid: 6C198FC83ABE
> [1650455846.949518][8339:8344] CHIP:TOO:         Channel: 44
> [1650455846.949540][8339:8344] CHIP:TOO:         WiFiBand: 2
> [1650455846.949561][8339:8344] CHIP:TOO:         Rssi: -43
> [1650455846.949583][8339:8344] CHIP:TOO:        }
> [1650455846.949614][8339:8344] CHIP:TOO:       [5]: {
> [1650455846.949637][8339:8344] CHIP:TOO:         Security: 8
> [1650455846.949659][8339:8344] CHIP:TOO:         Ssid: 47524C50726976617465
> [1650455846.949682][8339:8344] CHIP:TOO:         Bssid: C006C3F95F31
> [1650455846.949703][8339:8344] CHIP:TOO:         Channel: 48
> [1650455846.949725][8339:8344] CHIP:TOO:         WiFiBand: 2
> [1650455846.949747][8339:8344] CHIP:TOO:         Rssi: -51
> [1650455846.949768][8339:8344] CHIP:TOO:        }
> [1650455846.949799][8339:8344] CHIP:TOO:       [6]: {
> [1650455846.949823][8339:8344] CHIP:TOO:         Security: 8
> [1650455846.949846][8339:8344] CHIP:TOO:         Ssid: 47524C50726976617465
> [1650455846.949869][8339:8344] CHIP:TOO:         Bssid: C006C3F95EEB
> [1650455846.949891][8339:8344] CHIP:TOO:         Channel: 153
> [1650455846.949913][8339:8344] CHIP:TOO:         WiFiBand: 2
> [1650455846.949935][8339:8344] CHIP:TOO:         Rssi: -50
> [1650455846.949957][8339:8344] CHIP:TOO:        }
> [1650455846.949990][8339:8344] CHIP:TOO:       [7]: {
> [1650455846.950013][8339:8344] CHIP:TOO:         Security: 8
[... 53 lines cut ...]
> [1650455846.951263][8339:8344] CHIP:TOO:        }
> [1650455846.951293][8339:8344] CHIP:TOO:       [14]: {
> [1650455846.951315][8339:8344] CHIP:TOO:         Security: 12
> [1650455846.951337][8339:8344] CHIP:TOO:         Ssid: 7A69676265652D746872656164
> [1650455846.951360][8339:8344] CHIP:TOO:         Bssid: 6C198FC83ABC
> [1650455846.951381][8339:8344] CHIP:TOO:         Channel: 1
> [1650455846.951402][8339:8344] CHIP:TOO:         WiFiBand: 0
> [1650455846.951424][8339:8344] CHIP:TOO:         Rssi: -32
> [1650455846.951445][8339:8344] CHIP:TOO:        }
> [1650455846.951475][8339:8344] CHIP:TOO:       [15]: {
> [1650455846.951498][8339:8344] CHIP:TOO:         Security: 8
> [1650455846.951520][8339:8344] CHIP:TOO:         Ssid: 47524C50726976617465
> [1650455846.951542][8339:8344] CHIP:TOO:         Bssid: C006C3F95F30
> [1650455846.951563][8339:8344] CHIP:TOO:         Channel: 11
> [1650455846.951585][8339:8344] CHIP:TOO:         WiFiBand: 0
> [1650455846.951606][8339:8344] CHIP:TOO:         Rssi: -32
> [1650455846.951627][8339:8344] CHIP:TOO:        }
> [1650455846.951651][8339:8344] CHIP:TOO:    }
> [1650455846.951791][8339:8344] CHIP:DMG: ICR moving to [AwaitingDe]
> [1650455846.951884][8339:8344] CHIP:EM: Sending Standalone Ack for MessageCounter:4606188 on exchange 27909i

As I can verify the issue. I am closing this ticket.

Log for reference
[scan-networks_EP.txt](https://github.com/project-chip/connectedhomeip/files/8521476/scan-networks_EP.txt)

closed 2022-04-20 by Sivaramgrl