project-chip/connectedhomeip #15782: attributes are not updated when command is success - Network commissioning

Sivaramgrl, 2022-03-03:
**### Problem**
When a Command is executed and if it's a success, then the corresponding attributes need to be updated with their list and values.
but in the network commissioning the list or values are not getting updated.

**### Expected behavior:**
Example: When executing the AddOrUpdateWiFiNetwork command and if it's successful, then the Networks attribute list needs to be updated with the new entry. but the list is not getting updated. 
Similar issue with other commands and attributes in the network commissioning cluster

**### Actual behavior:**
Example: When executing the AddOrUpdateWiFiNetwork command and if it's successful, then the Networks attribute list needs to be updated with the new entry. 
Similar to other commands and attributes in the network commissioning cluster

**Commands: sudo ./chip-tool networkcommissioning add-or-update-wi-fi-network GRLPrivate_EXT matter123 0 1 0**
[1646296498.367293][3133:3138] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0005
[1646296498.367351][3133:3138] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Command 0x0000_0005
[1646296498.367429][3133:3138] CHIP:TOO:   NetworkConfigResponse: {
[1646296498.367475][3133:3138] CHIP:TOO:     **networkingStatus: 0**
[1646296498.367508][3133:3138] CHIP:TOO:     debugText:
[1646296498.367539][3133:3138] CHIP:TOO:    }

**Commands: sudo ./chip-tool networkcommissioning read networks 1 0**
[1646296600.694915][3151:3156] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0001DataVersion: 1863389603
[1646296600.695012][3151:3156] CHIP:TOO:   **Networks: 0 entries**
[1646296600.695113][3151:3156] CHIP:EM: Sending Standalone Ack for MessageCounter:15138151 on exchange 45779i

full log:
[attributes update.txt](https://github.com/project-chip/connectedhomeip/files/8176273/attributes.update.txt)

PSONALl, 2022-04-12:
This issue occurs because there is no List of Networks as mentioned in the spec is implemented in the code yet

Current code has a Network attribute with NetworkId and a boolean connected flag 

Code needs to maintain a list of such Networks

erjiaqing, 2022-04-13:
What is the platform of the server?

Since this data comes from platform driver, proving the platform will help identifying the issue.

cjandhyala, 2022-04-13:
@erjiaqing this is on raspi (linux) platform

erjiaqing, 2022-04-14:
I cannot reproduce on be946b04e4

```
In [15]: await devCtrl.SendCommand(1, 0, Clusters.GeneralCommissioning.Commands.ArmFailSafe(expiryLengthSeconds=360))
Out[15]: 
ArmFailSafeResponse(
│   errorCode=<CommissioningError.kOk: 0>,
│   debugText=''
)

In [16]: await devCtrl.SendCommand(1, 0, Clusters.NetworkCommissioning.Commands.AddOrUpdateWiFiNetwork(ssid=b'(my home wifi ssid)', credentials=b'(my home wifi password)'))
Out[16]: 
NetworkConfigResponse(
│   networkingStatus=<NetworkCommissioningStatus.kSuccess: 0>,
│   debugText=None,
│   networkIndex=0
)

In [19]: await devCtrl.ReadAttribute(1, [(0, Clusters.NetworkCommissioning)], returnClusterObject=True)
Out[19]: 
{
│   0: {
│   │   <class 'chip.clusters.Objects.NetworkCommissioning'>: NetworkCommissioning(
│   │   │   (data version)=3994585040,
│   │   │   maxNetworks=1,
│   │   │   networks=[
│   │   │   │   NetworkInfo(
│   │   │   │   │   networkID=b'(my home wifi ssid)',
│   │   │   │   │   connected=True
│   │   │   │   )
│   │   │   ],

```

erjiaqing, 2022-04-14:
Please note the failsafe timer and ensure it is not expired, for example, the config is reverted **as expected** if I did not issue the `Clusters.GeneralCommissioning.Commands.CommissioningComplete` command in a few minutes.

Please verify:
- You have armed failsafe before issue "add or update wifi network" command
- Failsafe timer is not expired or you have issued "commissioning complete" command after add or update network and before issuing the read request.

```
In [1]: await devCtrl.ReadAttribute(1, [(0, Clusters.NetworkCommissioning)], returnClusterObject=True)
2022-04-14 10:31:04 MATSU-X1C-DEBIAN chip.SC[176670] ERROR The device does not support GetClock_RealTimeMS() API. This will eventually result in CASE session setup failures.
Out[1]: 
{
│   0: {
│   │   <class 'chip.clusters.Objects.NetworkCommissioning'>: NetworkCommissioning(
│   │   │   (data version)=3994585040,
│   │   │   maxNetworks=1,
│   │   │   networks=[],

```

erjiaqing, 2022-04-14:
Also, from the timestamp of your output, it is very likely that this is caused by failsafe timeout

```
[1646296498.367539][3133:3138] CHIP:TOO: }

[1646296600.694915][3151:3156] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0001DataVersion: 1863389603
```

You issued the read request 100s after the add or update network request.

Sivaramgrl, 2022-04-19:
Thanks for sharing the input @erjiaqing,
Based on your comments, I worked on the Networks commissioning cluster commands 
Below is my observation:

### Used Commit id: 
commit 87938704ed93dde93e960605ae6a714923efa2e0
Date:   Mon Apr 18 18:59:53 2022 -0700

### Steps followed:
before the failsafe timeout, I send the add or update network request message. 

### Output:
As the sequence is executed before failsafe timeout, I was able to get the response as expected.
Confirmed the output by reading the Networks attribute. The list holds the SSID which is requested in add or update Network command. but still, the list is updated as the connected status is FALSE. Which is tracked in different bug #17106

### Logs:
[Full DUT report-NC.txt](https://github.com/project-chip/connectedhomeip/files/8511615/Full.DUT.report-NC.txt)
[Full TH report -NC.txt](https://github.com/project-chip/connectedhomeip/files/8511623/Full.TH.report.-NC.txt)

As the issue is resolved now. I am closing this ticket.

closed 2022-04-19 by Sivaramgrl