project-chip/connectedhomeip #15794: [TC-IDM-4.2]-While verifying the Activatation the Subscription request messages, Chip-Tool aborted with a "Segmentation fault"

Rajashreekalmane, 2022-03-03:
#### Problem
While verifying the response of the subscription request messages Chip-Tool aborted with a Segmentation fault.

Here is the example Execution Command: sudo ./chip-tool basic subscribe node-label 0 100 0 1 0

Setup:
Chip-Tool: RPI
DUT: RPI
Platform : IP and Wifi
Commit-ID:-883528dd3dadfcc665492be864866ad2e3889e16			

PFA log below:-
[Subscription log.txt](https://github.com/project-chip/connectedhomeip/files/8177731/Subscription.log.txt)

bzbarsky-apple, 2022-03-04:
@Rajashreekalmane Can you reproduce this reliably?  If so, can you please run this under a debugger and get a stack and any other relevant information?

And in general, running things under the debugger all the time so you can get stacks and other information if they crash might be helpful...

bzbarsky-apple, 2022-03-04:
Also, it's possible this is the crash that #15816 was fixing....

closed 2022-04-18 by Rajashreekalmane