project-chip/connectedhomeip #15818: M5 board doesn't show long pin code (again)

kean-apple, 2022-03-03:
#### Problem
<what's wrong or missing, please include any applicable:
Looks like M5 board is no longer printing long code again..this worked yesterday

SHA: 3b834619186a84db03281a5c76f236562db27ebd

E (10136) chip[SVR]: SetupQRCode: [MT:-24J042C00KA0648G00]
E (10136) chip[SVR]: Copy/paste the below URL in a browser to see the QR Code:
E (10146) chip[SVR]: https://dhrishi.github.io/connectedhomeip/qrcode.html?data=MT%3A-24J042C00KA0648G00
E (10156) chip[SVR]: Manual pairing code: [34970112332]

Expected to see both short and long code printed.

kean-apple, 2022-03-03:
[no long pin code on m5 board.txt](https://github.com/project-chip/connectedhomeip/files/8180359/no.long.pin.code.on.m5.board.txt)

dhrishi, 2022-03-03:
@kean-apple It's a feature ;-) See the [issue](https://github.com/project-chip/connectedhomeip/issues/15076) and it's [fix](https://github.com/project-chip/connectedhomeip/pull/15506) merged just sometime back

kean-apple, 2022-03-03:
ok got it. How do I set custom commissioning flow bit for M5 board to show the long code?

jadhavrohit924, 2022-03-04:
[long-manual-pairing-code-diff.txt](https://github.com/project-chip/connectedhomeip/files/8185346/long-manual-pairing-code-diff.txt)
By adding this diff you can commissioning flow to show the long code.

jadhavrohit924, 2022-03-04:
[long-manual-pairing-code-diff.txt](https://github.com/project-chip/connectedhomeip/files/8185370/long-manual-pairing-code-diff.txt)
@kean-apple By adding this diff you can commissioning flow to show the long code.

kean-apple, 2022-03-04:
thanks. I will close this issue since its expected

closed 2022-03-04 by kean-apple