project-chip/connectedhomeip #15822: Binding Cluster impl doesn't respect per-endpoint scoping of the cluster data

mrjerryjohns, 2022-03-03:
#### Problem

The state of a Binding cluster is unique to each endpoint it is on. However, the current Binding cluster implementation treats the state as shared across all endpoints.

This means that if I write to a Binding cluster on one endpoint, it over-writes data for the cluster on all other endpoints (and vice-versa).

Similarly, when reading out the Binding cluster data on one end point, it reads out data for all other endpoints as well.

This is not compliant to the spec.

#### Solution

`BindingTableAccess::WriteBindingTable` should only replace entries where the `path.mEndpointId` matches the `bindingTableIter->local`. 

`BindingTableAccess::ReadBindingTable` should only return entries where the `path.mEndpointId` matches the `entry.local`.

YAML tests should be added to validate the above behaviors.

mrjerryjohns, 2022-03-03:
@gjc13 do you have cycles this week to help address this?

gjc13, 2022-03-07:
Yes. Will create a PR to fix it.

closed 2022-03-08 by bzbarsky-apple