project-chip/connectedhomeip #15831: Update resource minimas for IM as per spec

mrjerryjohns, 2022-03-03:
#### Problem

Vet and update resource minimas in the IM as per spec.

mrjerryjohns, 2022-03-17:
Importantly, a couple of things have to happen:

1. Ensure that resource minimas for handling reads are preserved and not cannibalized by subscriptions.
2. Ensure that minimas for path handling is preserved.

mrjerryjohns, 2022-03-23:
A plausible algorithm that attempts to not be overly conservative about allotments while scaling correctly to actual # of fabrics/controllers interacting the device:

1. Always ensure we can at least hit the following minimas: at least 3 paths / sub,  at least 3 subs / fabric,  at least 5 fabrics by ensuring we either have `CHIP_SYSTEM_CONFIG_POOL_USE_HEAP` enabled, OR, the defines for `CHIP_IM_MAX_NUM_READ_HANDLER`, `CHIP_IM_SERVER_MAX_NUM_PATH_GROUPS`, are set correctly to reflect the above.
2. We divy up the available resources by the number of active fabrics and controllers on each fabric to compute the **maximum** available per subscription.

E.g we currently have 1 fabric commissioned. A subscription comes in (no other active subs). That subscription is allowed to use up all 3 X 3 X 5 = 35 paths. If a second subscription comes in, we now have to split the 35 paths in 2, so only 17 paths are permitted each. 

If a second fabric gets commissioned, we immediately divvy up the paths into two so that each fabric only gets 17 paths max. Which means the two existing subscriptions on the same fabric can only consume 8.5 paths each.

From a number of ReadHandlers perspectives, a similar algorithm is applied. Up to 3 subs X 5 fabrics = 15 handlers that can be consumed by a single fabric if only one fabric is present. If two fabrics are present, only 7.5 handlers are permitted each, etc.

mrjerryjohns, 2022-03-23:
We should reserve a separate pool for Reads exclusively: 1 ReadHandler / fabric, each that can support a minimum of 9 paths each.

So that would need a dedicated path pool of 9 x 5 = 35 as well.

mrjerryjohns, 2022-04-07:
So after thinking about this a bit more, I think we should pivot to a slightly different model that provides better utilization of these scant resources. The scheme I'm proposing is essentially a variation on the LRU scheme, i.e a "Contextual LRU":

1. Validate that an inbound request respects per-fabric minimas. If they do not AND we do not have any remaining space in the respective path pools, reject the subscription with 'PATHS_EXHAUSTED'.
2. Always permit a new request to succeed by evicting existing resources if needed to make way for the new interaction.
3. If there are no free `ReadHandler`s available, select a handler for eviction using the following policy:
    1. Find a fabric that is consuming more handlers than specified in the minima for per/fabric subscriptions/reads, then find the older handler within that fabric.
    2. If there are no overloaded fabrics, find the oldest handler in the accessing fabric.
    3. Evict handler.
4. Once a handler has been allocated, parse and allocate paths for the handler. If there are no free paths, select a handler for eviction using the same selection policy used above. Repeat until we have sufficient paths available to service the new request.

This is pretty much the same algorithm I'll be recommending for CASE session pool management as well.

mrjerryjohns, 2022-04-07:
Thanks @robszewczyk for helping with the algo above.

msandstedt, 2022-04-08:
@mrjerryjohns ,

I think this can work very well, and can address concerns about efficiency and unused resources.

I'm also concerned about these cases:

* single subscriber in a fabric: should have ability to consume all path resources allocated for that fabric
* multiple subscribers in a fabric: should have ability to consume at least the per-subscription path allocation

The reason for the former is that fabrics should be able to consume their maximum allocation in the way they choose.  The reason for the second point is that, as pointed out to me by @tcarmelveilleux , another valid use case is multiple controllers in a fabric that do not coordinate, and yet must still have guaranteed minimal available resources.

To handle this, I think we need to be more specific about eviction in point #4, and need to have both per-fabric path allocations and per-subscription path allocations.  With that, #4 expands to:

4. Once a handler has been allocated, parse and allocate paths for the handler. If there are no free paths, select a handler for eviction using this selection policy:
    1. Find a fabric that is consuming more subscription/read paths than specified in the minima for per/fabric, then find the oldest handler within that fabric (oldest = handler with the oldest timestamp at which the initial request had been serviced).
    2. If there are no over-committed fabrics, find a subscription/read handler in the current fabric that is consuming more paths than specified per subscription/read in the minima
    3. If there are no over-committed subscriptions/read handlers in the fabric, find the oldest handler in the accessing fabric.
    4. Evict handler.

mrjerryjohns, 2022-04-08:
Those suggestions make sense, will incorporate!

erjiaqing, 2022-04-11:
> @mrjerryjohns ,
>
> I think this can work very well, and can address concerns about efficiency and unused resources.
>
> I'm also concerned about these cases:
>
> * single subscriber in a fabric: should have [...]

@msandstedt I'm also thining about "Validate that an inbound request respects per-fabric minimas", What if we have 3 subscriptions with 3 paths each, then we received an incoming subscription request with 9 paths. Subscription with 9 paths does not exceeds the per fabric quota, however, it will kill all 3 existing subscriptions in the same fabric, which does not makes sense to me.

While the subscription will always success in this case (since we will evict other subscriptions in the same fabric), it might be an issue that the client may resubscribe forever, if the subscribed path is a wildcard path, or it contains a large array, then it will cause lots of traffic.

So I think 1 and 2 should also be modified:
- The first subscription of each fabric can use as many paths as possible within the fabric quota.
- The following subscriptions are only guranteed to use per subscription alloction
  - If they requests more than per subscription allocation, and we have enough resource for it, then it may proceed
  - Or we reject it with PATHS_EXHAUSTED

For example, say fabric hard upper bound is 9 paths, and subscription (soft) upper bound is 3 paths, then

```
Sub A: 5 paths - accept
Sub B: 4 paths - accpet
Sub C: 4 paths - reject, since we are running out of paths
Sub D: 3 paths - accpet, kill A
Sub E: 4 paths - reject
Sub F: 3 paths - accept, kill B
```

msandstedt, 2022-04-12:
> @msandstedt I'm also thining about "Validate that an inbound request respects per-fabric minimas", What if we have 3 subscriptions with 3 paths each, then we received an incoming subscription request with 9 paths. Subscription with 9 paths does not exceeds the per fabric quota, however, it will kill all 3 existing subscriptions in the same fabric, which does not makes sense to me.
> 
> While the subscription will always success in this case (since we will evict other subscriptions in the same fabric), it might be an issue that the client may resubscribe forever, if the subscribed path is a wildcard path, or it contains a large array, then it will cause lots of traffic.
> 
> So I think 1 and 2 should also be modified:
> 
> * The first subscription of each fabric can use as many paths as possible within the fabric quota.
> * Subscriptions are only guranteed iff the paths they used is no larger then per subscription alloction
>   
>   * If they requests more than per subscription allocation, and we have enough resource for it, then it may proceed
>   * Or we reject it with PATHS_EXHAUSTED

Yes, the 9-path subscription should not evict in-fabric subscriptions.  It should only be possible consume 9 paths if you are the first and only subscriber in your fabric.

That is tenable because a fabric can be administered such that you are guaranteed to only have one subscriber.  This can be accomplished by only allowing one reader in any of your deployed node ACLs.

So to summarize, the administrative proposition is this:

1. if you have multiple readers in your fabric, you must consider the per-subscription minima
2. if you have only one reader in your fabric, you may consider the per-fabric minima

PR #16811 (2022-04-25): [IM] Update resource minimas for IM as per spec
#### Problem

Fixes #15831 

#### Change overview
- Check the resource usage on read / subscription requests
- Reject the request with ResourceExhausted status code

#### Testing

Test w/ REPL

Introduced a new flag on build_examples.py (`test`) to force enable some logic.

Introduced a new build flag (`CHIP_CONFIG_IM_FORCE_FABRIC_QUOTA_CHECK=${chip_im_force_fabric_quota_check}`) to limit the resource usage even when the resources are allocated on heap.

- Try to establish as many subscriptions as possible on each fabric, expect success or resour [... cut]
closed 2022-04-25 by erjiaqing