project-chip/connectedhomeip #15845: TC-IDM-4.2]-Chip-tool needs to support the"  KeepSubscriptions flag" value to set.

Rajashreekalmane, 2022-03-04:
#### Problem
Chip-tool needs to provide a way to set the  KeepSubscriptions flag to set the value.
 As per the Test-Plan, the test step 16 & 17 in [TC-IDM-4.2] is expecting the keepsubscription flag to set true.

Test-Plan Reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-4-2-reportdatamessage-from-dut-test-cases

Rajashreekalmane, 2022-03-18:
[TC-IDM-4.1]Chip-Tool needs to provide a way to set the suppressresponse value for subscription request messages 

 PFA screenshot below
https://user-images.githubusercontent.com/95841834/158975391-37745ae8-8ca7-4dce-b709-a5fe8b1fdd9e.png

Garik098, 2022-03-23:
Need the same above mentioned support for chip-tool-darwin as well

vivien-apple, 2022-03-28:
@Rajashreekalmane The second request `[TC-IDM-4.1]Chip-Tool need to provide a way to set SuppressResponse field to True in step 10 for subscription request messages` is a separate issue. Can you file a separate issue for it ?
Furthermore I believe [TC-IDM-4.1] expect the test harness itself to answer with the `SuppressResponse` field set false which is an answer from the `Test Harness` not a command from `chip-tool`.

vivien-apple, 2022-03-28:
> 

> Need the same above mentioned support for chip-tool-darwin as well

That would be a separate issue as well. Can you file such an issue.

PR #16706 (2022-04-01): [chip-tool] Allow setting the KeepSubscriptions flag
#### Problem

`chip-tool` can not set the `KeepSubscriptions` flag.

fix #15845

#### Change overview
 * Add `chip-tool` support for setting the `KeepSubscriptions` flag.

#### Testing
I have checked the logs to make sure the flag is set appropriately.
closed 2022-04-01 by vivien-apple