project-chip/connectedhomeip #15861: M5 minimum connection time for wifi is too short

cecille, 2022-03-04:
#### Problem
M5 network cluster minimum connection time is set to 20s. We have definitely seen connections take longer than this.

 #### Proposed Solution
ESP platform devs would need to set this to the proper number.

cecille, 2022-03-07:
@g-coppock - can you give a summary of the scenario that was causing timeouts?

g-coppock, 2022-03-07:
I didn't save any measurements, but the original issue I'd seen was that the command timeout was defaulting to 12 seconds (which would always cause a failure if the first connect attempt failed). We had just changed the overall command timeout to 40 seconds temporarily.

cecille, 2022-03-07:
I think the failure happened when there was a retry on the connection.

I don't have any significant insight here - I might well have been wrong about 20s being too short. It was just lower than what we were using. Feel free to close if ESP is satisfied with the value as-is.

jadhavrohit924, 2022-03-08:
I’ll  put up a PR that bumps up the timeout to 30s.

PR #15948 (2022-03-08): [ESP32] Bump up wifi connection time
#### Problem
* Sometime wifi connection gets timeout.
* Fixes #15861 

#### Change overview
bumps up wifi connection time to 30s.

#### Testing
Commissioned all-clusters-app
closed 2022-03-08 by andy31415