project-chip/connectedhomeip #15892: Incorrect fabric ID validation in fabric table

bzbarsky-apple, 2022-03-05:
#### Problem
As far as I can tell, the fabric id validation added in #15829 does not match the spec.  It both allows things the spec does not allow, and disallows things the spec allows.

Specifically:
* If the ICAC has a fabric id of 0, that is invalid per spec, but that code allows it, as long as there is no fabric id in the RCAC.
* If the RCAC has a fabric id of 0, that is invalid per spec, but that code allows it.
* If the RCAC has a fabric ID but the ICAC does not, that is valid per spec as far as I can tell, but that code disallows it.

 #### Proposed Solution
Fix the validation.

emargolis, 2022-03-07:
I posted #15920 to address some of the comments in this ticket.

Note that checks for validity of the NodeId, FabricId and other DN attributes is performed when
loading certificates for validation (LoadCert()). So there is no need to check validity
again in FabricInfo::VerifyCredentials().

bzbarsky-apple, 2022-03-09:
If we do the checks elsewhere, great.  Looks like #15920 removes the checking for things that are not actually required to be true, so that would be sufficient to resolve this if the other checks are done elsewhere.

emargolis, 2022-03-10:
addressed with #15920

closed 2022-03-10 by emargolis