project-chip/connectedhomeip #15927: [TC-MF-1.2] Operational Credentials read fabrics doesn't return the full list of fabrics

mkardous-silabs, 2022-03-07:
#### Problem
When using this command `chip-tool operationalcredentials read fabrics 4057 0`, the returned list isn't complete

Controller 1
```
[1646683654.296924][2130:2135] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_003E Attribute 0x0000_0001DataVersion: 1261539983
[1646683654.297111][2130:2135] CHIP:TOO:   Fabrics: 1 entries
[1646683654.297251][2130:2135] CHIP:TOO:     [1]: {
[1646683654.297305][2130:2135] CHIP:TOO:       FabricIndex: 1
[1646683654.297363][2130:2135] CHIP:TOO:       RootPublicKey: 04EB049D6E18CCDFBDCEF78AC7548253188E3844CFEA04283ADEF51EBFFC440C3313458B6629A5C2A19EFDC8B64A24EBE2BF9C847FE504A454C3C0DEA7DD86511A
[1646683654.297417][2130:2135] CHIP:TOO:       VendorId: 65521
[1646683654.297464][2130:2135] CHIP:TOO:       FabricId: 1
[1646683654.297564][2130:2135] CHIP:TOO:       NodeId: 4057
[1646683654.297612][2130:2135] CHIP:TOO:       Label:
[1646683654.297660][2130:2135] CHIP:TOO:      }
```

Controller 2
```
[1646683258.826623][2291:2296] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_003E Attribute 0x0000_0001DataVersion: 1261539967
[1646683258.831173][2291:2296] CHIP:TOO:   Fabrics: 1 entries
[1646683258.831740][2291:2296] CHIP:TOO:     [1]: {
[1646683258.831831][2291:2296] CHIP:TOO:       FabricIndex: 2
[1646683258.831932][2291:2296] CHIP:TOO:       RootPublicKey: 049ED3B280B8EF33C35045B4E4EE1E114DAD9C18B32551584E15708115349960B467AA41DE724556C591B77EA9A4E7A789F091D4A9E48DCFFB7B4375B27283A24E
[1646683258.831993][2291:2296] CHIP:TOO:       VendorId: 65521
[1646683258.832044][2291:2296] CHIP:TOO:       FabricId: 1
[1646683258.832088][2291:2296] CHIP:TOO:       NodeId: 2
[1646683258.832136][2291:2296] CHIP:TOO:       Label:
[1646683258.832188][2291:2296] CHIP:TOO:      }
```

mrjerryjohns, 2022-03-08:
It's likely defaulting to doing fabric filtered reads.

@bzbarsky-apple would know how to set that option to false when doing reads to see the full contents of that list (modulo fabric-sensitive fields).

mkardous-silabs, 2022-03-08:
In TE8 channel, adding this flag is said to solve the problem ` --fabric-filtered 0`
Once i've had time to test the flag, i'll update the issue or close it accordingly .

bzbarsky-apple, 2022-03-09:
Yes, the default is fabric-filtered reads, and yes `--fabric-filtered 0` should help.

closed 2022-03-09 by mkardous-silabs