project-chip/connectedhomeip #15944: Chip CLI commissioning M5 fails with error "Failed to perform commissioning step 15"

vinay-apple, 2022-03-08:
#### Problem

Chip CLI commissioning M5 fails with error "Failed to perform commissioning step 15"

SHA: c0373bbc244894909fb0acf55c73444e14088208

Steps to reproduce:
1. Pair M5 with Chip CLI and observe that pairing starts but then it fails around connecting to network. 2. 

Chip CLI logs:

[1646708965038] [42868:841244] CHIP: [CTL] Failed to perform commissioning step 15
[1646708965038] [42868:841244] CHIP: [CTL] Rendezvous cleanup
[1646708965038] [42868:841244] CHIP: [TOO] Device commissioning Failure: ../../src/app/CommandSender.cpp:211: CHIP Error 0x00000032: Timeout
[1646708965038] [42868:841244] CHIP: [DMG] ICR moving to [AwaitingDe]
[1646708965038] [42868:841237] CHIP: [CTL] Shutting down the commissioner
[1646708965038] [42868:841237] CHIP: [CTL] Shutting down the controller
[1646708965038] [42868:841237] CHIP: [CTL] Shutting down the commissioner
[1646708965038] [42868:841237] CHIP: [CTL] Shutting down the controller
[1646708965038] [42868:841237] CHIP: [IN] Expiring all connections for fabric 1!!
[1646708965038] [42868:841237] CHIP: [CTL] Shutting down the commissioner
[1646708965038] [42868:841237] CHIP: [CTL] Shutting down the controller
[1646708965038] [42868:841237] CHIP: [IN] Expiring all connections for fabric 2!!
[1646708965038] [42868:841237] CHIP: [CTL] Shutting down the commissioner
[1646708965038] [42868:841237] CHIP: [CTL] Shutting down the controller
[1646708965038] [42868:841237] CHIP: [IN] Expiring all connections for fabric 3!!
[1646708965038] [42868:841237] CHIP: [CTL] Shutting down the System State, this will teardown the CHIP Stack
[1646708965038] [42868:841237] CHIP: [DMG] IM WH moving to [Uninitialized]
[1646708965038] [42868:841237] CHIP: [DMG] IM WH moving to [Uninitialized]
[1646708965038] [42868:841237] CHIP: [DMG] IM WH moving to [Uninitialized]
[1646708965038] [42868:841237] CHIP: [DMG] IM WH moving to [Uninitialized]
[1646708965039] [42868:841237] CHIP: [BLE] CancelConnection
[1646708965039] [42868:841237] CHIP: [IN] Clearing BLE pending packets.
[1646708965039] [42868:841237] CHIP: [BLE] BtpEngine unsub failed
[1646708965039] [42868:841237] CHIP: [BLE] Auto-closing end point's BLE connection.
[1646708965039] [42868:841237] CHIP: [DL] Inet Layer shutdown
[1646708965039] [42868:841237] CHIP: [DL] BLE shutdown
[1646708965039] [42868:841237] CHIP: [DL] System Layer shutdown
[1646708965039] [42868:841237] CHIP: [TOO] Run command failure: ../../src/app/CommandSender.cpp:211: CHIP Error 0x00000032: Timeout
[1646708965046] [42868:841237] CHIP: [SPT] VerifyOrDie failure at ../../src/lib/support/Pool.h:311: Allocated() == 0

M5 logs:

W (95338) wifi:Haven't to connect to a suitable AP now!
I (95338) chip[DL]: Attempting to connect WiFi station interface
I (95348) chip[DL]: WiFi station state change: NotConnected -> Connecting
I (95358) chip[DL]: Done driving station state, nothing else to do...
I (98478) chip[DL]: WIFI_EVENT_STA_DISCONNECTED
I (98478) chip[DL]: WiFi station state change: Connecting -> Connecting_Failed
W (98478) wifi:Haven't to connect to a suitable AP now!
I (98498) chip[DL]: WiFi station state change: Connecting_Failed -> NotConnected
I (98498) chip[DL]: Next WiFi station reconnect in 5000 ms
I (98508) chip[DL]: Done driving station state, nothing else to do...
I (98508) app-devicecallbacks: Current free heap: 68840

 #### Proposed Solution

[ChipCLI-logs-fails-commissioning.txt](https://github.com/project-chip/connectedhomeip/files/8202445/ChipCLI-logs-fails-commissioning.txt)
[M5-logs-chip-cli-fails-commissioning.txt](https://github.com/project-chip/connectedhomeip/files/8202446/M5-logs-chip-cli-fails-commissioning.txt)

vinay-apple, 2022-03-08:
Similar issue also seen when we try to commission Nordic board on thread network via chip tool cli. See attached logs. 

SHA: c0373bbc244894909fb0acf55c73444e14088208

[ChipCLI-Nordic-fails-commissioning.txt](https://github.com/project-chip/connectedhomeip/files/8207519/ChipCLI-Nordic-fails-commissioning.txt)
[Nordic-logs-fails-commissioning.txt](https://github.com/project-chip/connectedhomeip/files/8207520/Nordic-logs-fails-commissioning.txt)

bzbarsky-apple, 2022-05-21:
@vinay-apple Is this still being an issue?

ndyck14, 2022-09-16:
this issue is pretty old, but i ran into the same issue. dataset needs to be prefixed with `hex:` in the arguments. it should probably be added to the argument instructions

closed 2023-03-25 by stale[bot]