project-chip/connectedhomeip #15952: [ESP32] On-network commissioning does not work

jadhavrohit924, 2022-03-08:
#### Problem
* After boot device is not connecting to provided AP.
Through menuconfig
* Changed Rendezvous mode to On-Network ( Demo -->  Rendezvous Mode ).
* Set ssid and password  ( Example Connection Configuration ).
* Also tried with full IP-based commissioning ( Component config  --> CHIP Device Layer  --> Commissioning options  --> )

 #### Proposed Solution
After boot device should connet to AP

dhrishi, 2022-03-08:
@cecille Can you please check if we are missing something here?

jadhavrohit924, 2022-03-08:
Setting ssid and pass through ( Component config --> CHIP Device Layer --> WiFi Station Options ) is working.
And no need of full IP-based commissioning.

dhrishi, 2022-03-08:
Thanks @jadhavrohit924 . @cecille @wqx6 Please confirm what should be used here?

cecille, 2022-03-08:
I'm not sure what you're trying to do. If you're using on-network commissioning, the device should already be connected to the network before commissioning is attempted.

dhrishi, 2022-03-08:
@cecille That is right. The question really is about configuration of SSID and passphrase through menuconfig. And there seem to be 2 choices:

i. idf.py menuconfig -> Example Connection Configuration (with this on-network commissioning does not work)

ii. idf.py menuconfig -> Component config --> CHIP Device Layer --> WiFi Station Options (with this on-network commissioning works)

If i. above is to be used then commissioning should be fixed and ii. removed

If ii. above is to be used then the fix is not required and i. should be removed

cecille, 2022-03-09:
I've always used ii and it seems to work.

I also don't know what i was intended for. I think we need espressif folks to chime in on that one. I'm not sure who though, because I don't think that original option is from us. It's not in the chip Kconfig.

dhrishi, 2022-03-09:
@cecille I thought it was added by some Matter folks related to on-network commissioning and hence pulled you in. Sorry for the confusion :-)

Thanks a lot @chshu for the clarification.

dhrishi, 2022-03-11:
#15955 is merged. Closing this

closed 2022-03-11 by dhrishi