project-chip/connectedhomeip #15960: Power source cluster testcase [TC-PS-2.1] is failed during execution in master branch.

somu1710, 2022-03-08:
While executing power source cluster  test case [TC-PS-2.1] is failed during execution of chip-tool commands 

 * expected behavior - pass , status 0x0
 * actual behavior - failure , status - 0x86 , General error
failing commands - 
./chip-tool powersource read wired-assessed-input-voltage 1 1
  ./chip-tool powersource read wired-assessed-input-frequency 1 1
 ./chip-tool powersource read wired-current-type 1 1
./chip-tool powersource read wired-assessed-current 1 1
./chip-tool powersource read wired-nominal-voltage 1 1
./chip-tool powersource read wired-maximum-current 1 1
./chip-tool powersource read wired-present 1 1
./chip-tool powersource read active-wired-faults 1 1
./chip-tool powersource read battery-replacement-needed 1 1
./chip-tool powersource read battery-replaceability 1 1
./chip-tool powersource read battery-present 1 1
./chip-tool powersource read battery-replacement-description 1 1
./chip-tool powersource read battery-common-designation 1 1
./chip-tool powersource read battery-ansidesignation 1 1
./chip-tool powersource read battery-iecdesignation 1 1
./chip-tool powersource read battery-approved-chemistry 1 1
./chip-tool powersource read battery-capacity 1 1
./chip-tool powersource read battery-quantity 1 1
./chip-tool powersource read battery-time-to-full-charge 1 1
./chip-tool powersource read battery-functional-while-charging 1 1
./chip-tool powersource read battery-charging-current 1 1
./chip-tool powersource read active-battery-charge-faults 1 1

Here i attached the screenshot for your reference
[image]

And here I attached one log also
[PRS2.1 FAIL.txt](https://github.com/project-chip/connectedhomeip/files/8206433/PRS2.1.FAIL.txt)

andy31415, 2022-04-25:
I believe a lot of these attributes depend on features - like WIRED feature for the wired attbitues and `BAT` for the battery information.

Test cases should be updated to apply the test based on what the feature map says is supported.

somu1710, 2022-04-29:
As feature map supports for the WIRED feature ,then it should be implement right ?

FYI-**./chip-tool powersource read feature-map 1 1**

[1651245038.584442][2428:2433] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_002F Attribute 0x0000_FFFC DataVersion: 1887093579
[1651245038.584599][2428:2433] CHIP:TOO:   **FeatureMap: 0**

[image]

andy31415, 2022-05-09:
I see featuremap is 0, so no features are enabled/supported.

I expect bitwise to have 0x01 to be wired, 0x02 to be bat, 0x04 to be rech and 0x08 to be replc.

To be specific: "Feature Map == 0" means no features (the list of features mentions "bits" so bit 0 is 0x01 ... it would have been clearer if spec would have written the constant bitflag as well I guess).

andy31415, 2022-05-09:
> some of the attributes in BAT features are mentioned as mandatory in test plan but as per spec in the conformance its mentioned as [BAT].

Spec is always considered correct in my understanding. Test plan should be adjusted and remove the mandatory bits if spec says optional.

somu1710, 2022-05-10:
Then in the case test plan needs to update as per spec right? If it changes means we cant able to verify this test case because of no features.This test case is only verifiyable by using real DUT which is supports the features?

somu1710, 2022-05-11:
In recent commit its fixed and the feature map value is getting as 2 and its supporting BAT mandatory attributes.so i am closing the issue.

closed 2022-05-11 by somu1710