project-chip/connectedhomeip #15969: iOS Chiptool- Failed to perform commissioning step 12 CHIP Error 0x0000000B: No memory

kean-apple, 2022-03-08:
#### Problem
After attempting to pair M5 board with CHIPtool multiple times, get into state where CHIP Error 0x0000000B: No memory

initial: 2 chip tools app on same iOS device
1. Pair M5 board using iOS chiptool 
2. Toggle light 3. 
3. unpair & reset M5 board
4. repeat steps 1-3 a few times with each Chiptool app

After repeating pairing a few times eg. 10 times (including failures), both chip tool get into a start where pairing never starts due to CHIP Error 0x0000000B: No memory

2022-03-08 09:15:22.509970-0800  localhost CHIPTool[20011]: (CHIP) [com.zigbee.chip:all] ðµ [1646759722509] [20011:595436] CHIP: [CTL] Device returned status 5 on receiving the NOC
2022-03-08 09:15:22.510549-0800  localhost CHIPTool[20011]: (CHIP) [com.zigbee.chip:all] ðµ [1646759722510] [20011:595436] CHIP: [CTL] Add NOC failed with error ../../../../../../../../../../../connectedhomeip/src/controller/CHIPDeviceController.cpp:1228: CHIP Error 0x0000000B: No memory
2022-03-08 09:15:22.510647-0800  localhost CHIPTool[20011]: (CHIP) [com.zigbee.chip:all] ð´ [1646759722510] [20011:595436] CHIP: [CTL] Failed to perform commissioning step 12
2022-03-08 09:15:22.510810-0800  localhost CHIPTool[20011]: (CHIP) [com.zigbee.chip:all] ðµ [1646759722510] [20011:595436] CHIP: [CTL] Rendezvous cleanup
2022-03-08 09:15:22.510917-0800  localhost CHIPTool[20011]: (CHIP) DevicePairingDelegate Commissioning complete. NodeId 4 Status ../../../../../../../../../../../connectedhomeip/src/controller/CHIPDeviceController.cpp:1228: CHIP Error 0x0000000B: No memory

 #### Proposed Solution
<suggested fix, suggested enhancement>
[scanning-QR-code-no-memory.txt](https://github.com/project-chip/connectedhomeip/files/8208345/scanning-QR-code-no-memory.txt)

kean-apple, 2022-03-08:
I was finally able to get out of this state after updating my iOS device and re-installing CHIPtool. Removing smoke test blocker for now since I don't have great repro steps. I was hitting frequent chiptool crashes eg. https://github.com/project-chip/connectedhomeip/issues/15972 leading up to getting into this state

bzbarsky-apple, 2022-05-21:
@kean-apple Did you ever run into this again?

closed 2022-11-26 by stale[bot]

justinsignify494, 2024-08-26:
How to fix this problem without re-installing chip-tool?