project-chip/connectedhomeip #15972: iOS chiptool crashes when scanning QR code -

kean-apple, 2022-03-08:
#### Problem
iOS chiptool crashes when scanning QR code

SHA: 2822511e632bd2ae8a2e5a853a114a9bac1cebe5
1. Launch iOS chiptool
2. Launch QRcode scanner to pair a M5 board

In step 2, iOS chiptool crashes sometimes eg. 20
[iOSchiptool-crash-scanning-qr-code.txt](https://github.com/project-chip/connectedhomeip/files/8208648/iOSchiptool-crash-scanning-qr-code.txt)
%

kean-apple, 2022-03-08:
[image]

bzbarsky-apple, 2022-03-09:
@kean-apple So this is starting from a state when we have nothing paired and it crashes when you try to start the camera?

kean-apple, 2022-03-10:
Its generally happens after a few pairing attempts. For the most part I do unpair before attempting pairing but there might be some cases where there is a stale pairing before launching the camera..I'll try to characterize some more

kean-apple, 2022-03-11:
Actually, I feel like I see this crash more often if I use PIN code to start pairing then try scanning QR code

kean-apple, 2022-04-21:
Closing this since its not seen in a while

closed 2022-04-21 by kean-apple